RT-Thread/rt-thread #1410: rt-thread 使用c++ 编译遇到问题

chenxiuqimr, 2018-05-06:
测试了bsp下的stm32f407和stm32f103两个工程，env 里打开了对c++的支持项后保存配置
1、直接scons编译 出错
> scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\main.o
CC build\drivers\board.o
CC build\drivers\drv_usart.o
CC build\drivers\stm32f4xx_it.o
CXX build\kernel\components\cplusplus\crt.o
Error in calling:
g++ -o "build\kernel\components\cplusplus\crt.o" -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -ffunction-sections -fdata-sections -std=c99 -Dgcc -O0 -gdwarf-2 -g -D_REENT_SMALL -DSTM32F411xE -DUSE_HAL_DRIVER -DRT_USING_MINILIBC -I. -Iapplications -Idrivers -ILibraries\CMSIS\Device\ST\STM32F4xx\Include -ILibraries\CMSIS\Include -ILibraries\STM32F4xx_HAL_Driver\Inc -ID:\Project\RTT\rtt_aa\include -ID:\Project\RTT\rtt_aa\libcpu\arm\cortex-m4 -ID:\Project\RTT\rtt_aa\libcpu\arm\common -ID:\Project\RTT\rtt_aa\components\cplusplus -ID:\Project\RTT\rtt_aa\components\drivers\include -ID:\Project\RTT\rtt_aa\components\drivers\include -ID:\Project\RTT\rtt_aa\components\finsh -ID:\Project\RTT\rtt_aa\components\libc\compilers\minilibc "D:\Project\RTT\rtt_aa\components\cplusplus\crt.cpp"
Exception: [Error 2] : No such file or directory
scons: *** [build\kernel\components\cplusplus\crt.o] Error 2
scons: building terminated because of errors.

2、先用 scons --target=mdk5 生成mdk5工程，然后打开mdk5工程点击编译能编译成功，但是有两个警告
linking...
.\build\rtthread-stm32f4xx.axf: Warning: L6330W: Undefined symbol SHT$$INIT_ARRAY$$Base (referred from crt_init.o). Unused section has been removed.
.\build\rtthread-stm32f4xx.axf: Warning: L6330W: Undefined symbol SHT$$INIT_ARRAY$$Limit (referred from crt_init.o). Unused section has been removed.
Program Size: Code=56632 RO-data=4796 RW-data=436 ZI-data=4252  
Finished: 0 information, 2 warning and 0 error messages.
After Build - User command #1: fromelf --bin .\build\rtthread-stm32f4xx.axf --output rtthread.bin
".\build\rtthread-stm32f4xx.axf" - 0 Error(s), 2 Warning(s).

我想要c++进行开发，遇到以上问题，是不是有哪些地方没注意到呢？有无人用c++编译成功呢？

TanekLiang, 2018-05-06:
你可以参考一些打开c++的bsp，例如stm32h743-nucleo， i.mx rt 1052等等bsp。
需要修改的地方大致有：rtconfig.py里添加CXX和CXXFLAGS，SConstruct里添加CXX的信息。如果你在C里使用C99，还有一些需要注意的地方（参考i.mx rt 1052）

chenxiuqimr, 2018-05-07:
在bsp下找不到打开c++的demo呢，试了几个stm32的工程，打开了对c++的支持项RT_USING_CPLUSPLUS和RT_USING_LIBC，mdk5.21编译都提示这两个警告

.\build\app.axf: Warning: L6330W: Undefined symbol SHT$$INIT_ARRAY$$Base (referred from crt_init.o). Unused section has been removed.
.\build\app.axf: Warning: L6330W: Undefined symbol SHT$$INIT_ARRAY$$Limit (referred from crt_init.o). Unused section has been removed.

closed 2018-05-15 by chenxiuqimr

cl1217982730, 2019-10-29:
情况SHT$$INIT_ARRAY$$Base 这个warning 是怎么解决的？