RT-Thread/rt-thread #1446: bsp\imxrt1050-evk编译错误 in gcc

bjrobertshen, 2018-05-18:
新下载的RT with micropython。在编译时出错，好像同#1021的问题一样=》https://github.com/RT-Thread/rt-thread/issues/1021

错误如下：
`robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/Libraries/drivers/fsl_adc.o
CC build/Libraries/drivers/fsl_adc_etc.o
CC build/Libraries/drivers/fsl_aipstz.o
CC build/Libraries/drivers/fsl_aoi.o
CC build/Libraries/drivers/fsl_bee.o
CC build/Libraries/drivers/fsl_cache.o
CC build/Libraries/drivers/fsl_clock.o
CC build/Libraries/drivers/fsl_cmp.o
CC build/Libraries/drivers/fsl_common.o
CC build/Libraries/drivers/fsl_csi.o
CC build/Libraries/drivers/fsl_dcdc.o
CC build/Libraries/drivers/fsl_dcp.o
CC build/Libraries/drivers/fsl_dmamux.o
CC build/Libraries/drivers/fsl_edma.o
CC build/Libraries/drivers/fsl_elcdif.o
CC build/Libraries/drivers/fsl_enc.o
CC build/Libraries/drivers/fsl_enet.o
CC build/Libraries/drivers/fsl_ewm.o
CC build/Libraries/drivers/fsl_flexcan.o
CC build/Libraries/drivers/fsl_flexio.o
CC build/Libraries/drivers/fsl_flexio_i2c_master.o
CC build/Libraries/drivers/fsl_flexio_i2s.o
CC build/Libraries/drivers/fsl_flexio_i2s_edma.o
CC build/Libraries/drivers/fsl_flexio_spi.o
CC build/Libraries/drivers/fsl_flexio_spi_edma.o
CC build/Libraries/drivers/fsl_flexio_uart.o
CC build/Libraries/drivers/fsl_flexio_uart_edma.o
CC build/Libraries/drivers/fsl_flexram.o
CC build/Libraries/drivers/fsl_flexspi.o
CC build/Libraries/drivers/fsl_gpc.o
CC build/Libraries/drivers/fsl_gpio.o
CC build/Libraries/drivers/fsl_gpt.o
CC build/Libraries/drivers/fsl_kpp.o
CC build/Libraries/drivers/fsl_lpi2c.o
CC build/Libraries/drivers/fsl_lpi2c_edma.o
CC build/Libraries/drivers/fsl_lpspi.o
CC build/Libraries/drivers/fsl_lpspi_edma.o
CC build/Libraries/drivers/fsl_lpuart.o
CC build/Libraries/drivers/fsl_lpuart_edma.o
CC build/Libraries/drivers/fsl_pit.o
CC build/Libraries/drivers/fsl_pmu.o
CC build/Libraries/drivers/fsl_pwm.o
CC build/Libraries/drivers/fsl_pxp.o
CC build/Libraries/drivers/fsl_qtmr.o
CC build/Libraries/drivers/fsl_rtwdog.o
CC build/Libraries/drivers/fsl_sai.o
CC build/Libraries/drivers/fsl_sai_edma.o
CC build/Libraries/drivers/fsl_semc.o
CC build/Libraries/drivers/fsl_snvs_hp.o
CC build/Libraries/drivers/fsl_snvs_lp.o
CC build/Libraries/drivers/fsl_spdif.o
CC build/Libraries/drivers/fsl_spdif_edma.o
CC build/Libraries/drivers/fsl_src.o
CC build/Libraries/drivers/fsl_trng.o
CC build/Libraries/drivers/fsl_tsc.o
CC build/Libraries/drivers/fsl_usdhc.o
CC build/Libraries/drivers/fsl_wdog.o
CC build/Libraries/drivers/fsl_xbara.o
CC build/Libraries/drivers/fsl_xbarb.o
CC build/applications/device_test.o
CC build/applications/main.o
CC build/applications/mem_dump.o
CC build/applications/mem_test.o
CC build/drivers/board.o
CC build/drivers/drv_cache.o
CC build/drivers/drv_eth.o
CC build/drivers/drv_i2c.o
CC build/drivers/drv_lcd.o
CC build/drivers/drv_pin.o
drivers/drv_pin.c: In function 'rt1052_get_irqnum':
drivers/drv_pin.c:423:12: warning: 'irq_num' may be used uninitialized in this function [-Wmaybe-uninitialized]
     return irq_num;
            ^~~~~~~
CC build/drivers/drv_rtc.o
CC build/drivers/drv_sdio.o
CC build/drivers/drv_sdram.o
CC build/drivers/drv_spi_bus.o
CC build/drivers/drv_uart.o
CC build/drivers/fsl_phy.o
CC build/kernel/components/dfs/filesystems/devfs/devfs.o
CC build/kernel/components/dfs/filesystems/elmfat/dfs_elm.o
CC build/kernel/components/dfs/filesystems/elmfat/ff.o
CC build/kernel/components/dfs/filesystems/elmfat/option/ccsbcs.o
CC build/kernel/components/dfs/filesystems/net/dfs_net.o
CC build/kernel/components/dfs/filesystems/net/net_netdb.o
CC build/kernel/components/dfs/filesystems/net/net_sockets.o
CC build/kernel/components/dfs/src/dfs.o
CC build/kernel/components/dfs/src/dfs_file.o
CC build/kernel/components/dfs/src/dfs_fs.o
CC build/kernel/components/dfs/src/dfs_posix.o
CC build/kernel/components/dfs/src/poll.o
CC build/kernel/components/dfs/src/select.o
In file included from /home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:29:0:
/home/robert/mydisk/code/rt-thread/components/dfs/include/dfs_select.h:10:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                      ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/include/dfs_select.h:10:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                       ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/include/dfs_select.h:10:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                                         ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:31:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                      ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:31:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                                       ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:31:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                                                         ^~~~~~
scons: *** [build/kernel/components/dfs/src/select.o] Error 1
scons: building terminated because of errors.
robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$ 
`

TanekLiang, 2018-05-18:
你用的是哪个编译器？

bjrobertshen, 2018-05-18:
是这个（MCUExpresso 带的）：
 /usr/local/mcuxpressoide-10.1.1_606/ide/tools/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$

TanekLiang, 2018-05-18:
嗯，这个是编译器版本不适配的原因，你可以使用env里面自带的gcc。

bjrobertshen, 2018-05-18:
换了一个：
robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$ /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

上个编译错误解决了， 但在编译Modbus出错了：

robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$scons
......
CC build/kernel/components/libc/compilers/newlib/libc.o
/home/robert/mydisk/code/rt-thread/components/libc/compilers/newlib/libc.c: In function 'libc_system_init':
/home/robert/mydisk/code/rt-thread/components/libc/compilers/newlib/libc.c:55:5: warning: implicit declaration of function 'putenv' [-Wimplicit-function-declaration]
     putenv("PATH=/bin");
     ^
CC build/kernel/components/libc/compilers/newlib/libc_syms.o
CC build/kernel/components/libc/compilers/newlib/stdio.o
/home/robert/mydisk/code/rt-thread/components/libc/compilers/newlib/stdio.c: In function 'libc_stdio_set_console':
/home/robert/mydisk/code/rt-thread/components/libc/compilers/newlib/stdio.c:84:5: warning: implicit declaration of function 'fileno' [-Wimplicit-function-declaration]
     if (std_console) return fileno(std_console);
     ^
CC build/kernel/components/libc/compilers/newlib/syscalls.o
CC build/kernel/components/libc/compilers/newlib/time.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbfunccoils.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbfuncdiag.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbfuncdisc.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbfuncholding.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbfuncinput.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbfuncother.o
CC build/kernel/components/net/freemodbus/modbus/functions/mbutils.o
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/modbus/functions/mbutils.c: In function 'xMBUtilSetBits':
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/modbus/functions/mbutils.c:56:5: warning: implicit declaration of function 'assert_param' [-Wimplicit-function-declaration]
     assert_param( ucNBits <= 8 );
     ^
CC build/kernel/components/net/freemodbus/modbus/mb.o
CC build/kernel/components/net/freemodbus/modbus/rtu/mbcrc.o
CC build/kernel/components/net/freemodbus/modbus/rtu/mbrtu.o
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/modbus/rtu/mbrtu.c: In function 'eMBRTUReceive':
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/modbus/rtu/mbrtu.c:156:5: warning: implicit declaration of function 'assert_param' [-Wimplicit-function-declaration]
     assert_param( usRcvBufferPos < MB_SER_PDU_SIZE_MAX );
     ^
CC build/kernel/components/net/freemodbus/port/port.o
CC build/kernel/components/net/freemodbus/port/portevent.o
CC build/kernel/components/net/freemodbus/port/portserial.o
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/port/portserial.c: In function 'xMBPortSerialInit':
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/port/portserial.c:59:17: error: 'MODBUS_SLAVE_RT_CONTROL_PIN_INDEX' undeclared (first use in this function)
     rt_pin_mode(MODBUS_SLAVE_RT_CONTROL_PIN_INDEX, PIN_MODE_OUTPUT);
                 ^
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/port/portserial.c:59:17: note: each undeclared identifier is reported only once for each function it appears in
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/port/portserial.c: In function 'vMBPortSerialEnable':
/home/robert/mydisk/code/rt-thread/components/net/freemodbus/port/portserial.c:130:22: error: 'MODBUS_SLAVE_RT_CONTROL_PIN_INDEX' undeclared (first use in this function)
         rt_pin_write(MODBUS_SLAVE_RT_CONTROL_PIN_INDEX, PIN_LOW);
                      ^
scons: *** [build/kernel/components/net/freemodbus/port/portserial.o] Error 1
scons: building terminated because of errors.
robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$

TanekLiang, 2018-05-18:
MODBUS_SLAVE_RT_CONTROL_PIN_INDEX 没有声明，你可以看看为什么出错了。

bjrobertshen, 2018-05-18:
刚开始用 RT， 还真不知道在那个文件里设置pin config。能给些提示吗？

bjrobertshen, 2018-05-18:
menuconfig 删除了Modbus， 而在micropython的编译中还是出错了：

.....CC build/packages/micropython/port/machine_hw_i2c.o
CC build/packages/micropython/port/machine_hw_spi.o
packages/micropython/port/machine_hw_spi.c:117:13: warning: 'machine_hard_spi_deinit' defined but not used [-Wunused-function]
 STATIC void machine_hard_spi_deinit(mp_obj_base_t *self_in) {
             ^
CC build/packages/micropython/port/machine_pin.o
CC build/packages/micropython/port/machine_uart.o
In file included from packages/micropython/py/mpstate.h:35:0,
                 from packages/micropython/py/runtime.h:29,
                 from packages/micropython/port/machine_uart.c:30:
packages/micropython/port/machine_uart.c:216:29: error: static declaration of 'machine_uart_locals_dict' follows non-static declaration
 STATIC MP_DEFINE_CONST_DICT(machine_uart_locals_dict, machine_uart_locals_dict_table);
                             ^
packages/micropython/py/obj.h:309:25: note: in definition of macro 'MP_DEFINE_CONST_DICT'
     const mp_obj_dict_t dict_name = { \
                         ^
In file included from packages/micropython/port/machine_uart.c:36:0:
packages/micropython/port/machine_uart.h:34:28: note: previous declaration of 'machine_uart_locals_dict' was here
 extern const mp_obj_dict_t machine_uart_locals_dict;
                            ^
packages/micropython/port/machine_uart.c: In function 'machine_uart_ioctl':
packages/micropython/port/machine_uart.c:235:5: warning: return makes integer from pointer without a cast
     return NULL;
     ^
scons: *** [build/packages/micropython/port/machine_uart.o] Error 1
scons: building terminated because of errors.
robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$

TanekLiang, 2018-05-18:
@armink

armink, 2018-05-19:
@bjrobertshen 

- modbus 移植可以参考下[这里](https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/blob/b473707f3e7eb3cbaefbaeb05ff68a1d93de362a/BSP/inc/bsp.h#L57)。关于 Pin 设备，最好也阅读下 RTT 官方的 [《通用GPIO设备应用笔记》](https://www.rt-thread.org/document/site/rtthread-application-note/driver/gpio/an0002-rtthread-driver-gpio/
)
-MicroPython 在 GCC 下编译出错了问题 [修复了](https://www.rt-thread.org/document/site/rtthread-application-note/driver/gpio/an0002-rtthread-driver-gpio/) ，你再更新下软件包试试。

uestczyh222, 2018-05-19:
fix了

bjrobertshen, 2018-05-19:
Thanks!, the compiler is OK, but, the link is fail. 可能还是tools的版本问题。 在哪里可以下载到linux下的正确编译工具（不是windows下的）？

CC Libraries/system_MIMXRT1052.o
LINK rtthread-imxrt.elf
/opt/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: error: /opt/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o): Conflicting CPU architectures 13/1
/opt/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /opt/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o)
collect2: error: ld returned 1 exit status
scons: *** [rtthread-imxrt.elf] Error 1
scons: building terminated because of errors.

TanekLiang, 2018-05-19:
这个应该是工具链太旧的缘故，可以试试5.4版本

bjrobertshen, 2018-05-19:
OK. 刚下栽了最新的arm tools。 build is OK！ 谢谢！

arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bjrobertshen, 2018-05-20:
gpio输入还是有问题：用micropython的gpio设置SW8按钮为上拉输入。只要开始输入为高，按下sw8，变为低，抬起按钮输入应该变回高，但仍为低。只有重启，才会回到高电平。感觉没有上拉

armink, 2018-05-21:
@bjrobertshen 这个上手指南你那能跑吗？ https://www.rt-thread.org/document/site/rtthread-development-guide/micropython/docs/01-Getting_Started_Guide/#32

bjrobertshen, 2018-05-21:
能跑，就是用这个测试的。第一次按开关，变为低电平，LED灯灭；抬起按钮，灯也不亮，测量SW8就是低电平。该板子，跑nxp的input例程也是这样，昨天更新了最新的MCUXpresso，用pin config tool从新生成了pin_mux.c发现sw8的io设置多了一条IOMUXC_SetPinConfig(GPIO_IO00, 0x80B0u);的设置，以前没有。编译运行sw8按钮就正常了。我怀疑你们的config也要更新一下才行。

liu2guang, 2018-05-21:
@bjrobertshen 请问SW8是哪个IO口呢。

bjrobertshen, 2018-05-21:
检查driver下的drv_pin.c里的static void rt1052_pin_mode(rt_device_t dev, rt_base_t pin, rt_base_t mode)中设置PIN_MODE_INPUT_PULLUP的值：
        case PIN_MODE_INPUT_PULLUP:
        {
            config_value = 0x5030U;
            gpio.direction = kGPIO_DigitalInput;
        }
好像有问题。在rt1050的手册中0x400A8018（SW8用的pin）的设置, 你这里用的是0x5030；我改为0x7030就好了。

liu2guang, 2018-05-21:
@bjrobertshen ok我晚上下班的时候看一下。

armink, 2018-05-26:
@liu2guang 问题已确认？确认后 isuess 就可以关闭了。

bjrobertshen, 2018-05-26:
OK