RT-Thread/rt-thread #1492: LWIP1.4.1 TCP回环测试发现死锁在内部同步信号量上

geniusgogo, 2018-05-31:
我用stm32f207做tcpclient回环测试，5ms一个包，一个包1024byte。几十个包之后就会wait_forever在这个apimsg->msg.conn->op_completed信号量上。
但是能力有限不知道怎么进一步跟踪。
lwip_recvfrom->netconn_recved->tcpip_apimsg->sys_arch_sem_wait(&apimsg->msg.conn->op_completed, 0);

BernardXiong, 2018-05-31:
回环测试是用的loop back模式？切换到lwip 2.0.2，情况怎么样？

geniusgogo, 2018-05-31:
应用层自己做的回环上位机TCP SERVER，主动每隔5ms/1024byte.下位机TCP client单线程先阻塞recv，一次recv传参size=1024byte，然后收到多少就send出去。
`while (1)
        {
            r_size = recv(sock, r_buf, sizeof(r_buf), 0);
            // rt_kprintf("recv %d bytes\n", r_size);

            if (r_size > 0)
            {
                r_total += r_size;
                r_size = send(sock, r_buf, r_size, 0);

                if (r_size > 0)
                {
                    s_total += r_size;
                    // rt_kprintf("send total:%u bytes, read total:%u bytes\n", s_total, r_total);
                }
                else
                {
                    rt_kprintf("send failure:%d, errcode:%d\n", r_size, rt_get_errno());
                    lwip_close(sock);
                    break;
                }
            }
            else
            {
                rt_kprintf("recv failure:%d, errcode:%d\n", r_size, rt_get_errno());
                lwip_close(sock);
                break;
            }
        }`

2.0.2基于想同的配置，想同的应用测试代码。5ms/1024byte，一样几十个包就立马挂。

geniusgogo, 2018-05-31:
贴一下我的配置：
`#ifndef RT_CONFIG_H__
#define RT_CONFIG_H__

/* Automatically generated file; DO NOT EDIT. */
/* RT-Thread Project Configuration */

/* RT-Thread Kernel */

#define RT_NAME_MAX 6
#define RT_ALIGN_SIZE 4
#define RT_THREAD_PRIORITY_8
#define RT_THREAD_PRIORITY_MAX 8
#define RT_TICK_PER_SECOND 200
#define RT_DEBUG
#define RT_USING_OVERFLOW_CHECK
#define RT_DEBUG_INIT 0
#define RT_DEBUG_THREAD 0
#define IDLE_THREAD_STACK_SIZE 384

/* Inter-Thread communication */

#define RT_USING_SEMAPHORE
#define RT_USING_MUTEX
#define RT_USING_EVENT
#define RT_USING_MAILBOX

/* Memory Management */

#define RT_USING_SMALL_MEM
#define RT_USING_HEAP

/* Kernel Device Object */

#define RT_USING_DEVICE
#define RT_USING_CONSOLE
#define RT_CONSOLEBUF_SIZE 128
#define RT_CONSOLE_DEVICE_NAME "uart4"
#define ARCH_ARM
#define ARCH_ARM_CORTEX_M
#define ARCH_ARM_CORTEX_M3

/* RT-Thread Components */

/* C++ features */

/* Command shell */

/* Device virtual file system */

/* Device Drivers */

#define RT_USING_DEVICE_IPC

/* Using USB */

/* POSIX layer and C standard library */

/* Network stack */

/* light weight TCP/IP stack */

#define RT_USING_LWIP
#define RT_USING_LWIP202
#define RT_LWIP_ICMP

/* Static IPv4 Address */

#define RT_LWIP_IPADDR "192.168.1.30"
#define RT_LWIP_GWADDR "192.168.1.1"
#define RT_LWIP_MSKADDR "255.255.255.0"
#define RT_LWIP_UDP
#define RT_LWIP_TCP
#define RT_MEMP_NUM_NETCONN 2
#define RT_LWIP_PBUF_NUM 10
#define RT_LWIP_RAW_PCB_NUM 2
#define RT_LWIP_UDP_PCB_NUM 2
#define RT_LWIP_TCP_PCB_NUM 2
#define RT_LWIP_TCP_SEG_NUM 40
#define RT_LWIP_TCP_SND_BUF 8192
#define RT_LWIP_TCP_WND 8192
#define RT_LWIP_TCPTHREAD_PRIORITY 3
#define RT_LWIP_TCPTHREAD_MBOX_SIZE 4
#define RT_LWIP_TCPTHREAD_STACKSIZE 1024
#define RT_LWIP_ETHTHREAD_PRIORITY 5
#define RT_LWIP_ETHTHREAD_STACKSIZE 512
#define RT_LWIP_ETHTHREAD_MBOX_SIZE 4
#define LWIP_NETIF_STATUS_CALLBACK 1
#define SO_REUSE 0
#define LWIP_SO_RCVTIMEO 1
#define LWIP_SO_SNDTIMEO 1
#define LWIP_SO_RCVBUF 1
#define LWIP_NETIF_LOOPBACK 0

/* Modbus master and slave stack */

/* VBUS(Virtual Software BUS) */

/* Utilities */

/* ARM CMSIS */

/* RT-Thread online packages */

/* system packages */

/* RT-Thread GUI Engine */

/* IoT - internet of things */

/* Wi-Fi */

/* Marvell WiFi */

/* Wiced WiFi */

/* security packages */

/* language packages */

/* multimedia packages */

/* tools packages */

/* miscellaneous packages */

/* example package: hello */

#define SOC_STM32F2

#endif
`

BernardXiong, 2018-05-31:
好像没什么好办法，需要你自己多测试，多发现问题

closed 2022-08-06 by mysterywolf