RT-Thread/rt-thread #1504: Naming conflict: rt-thread's mem.c conflicts with lwip's mem.c

pansila, 2018-06-05:
rt-thread的mem.c和lwip的mem.c名字冲突，现在rt-thread的做法是用rt-thread的mem替代lwip的，但对于我们要做多个os兼容的场景，还是保留lwip的实现比较好，能否改名？或者有其它编译器选项绕过的办法让两者共存？

pansila, 2018-06-05:
三种工具链都会用到，具体怎么做？

TanekLiang, 2018-06-05:
默认情况，scons就启用了VariantDir，也就是编译的中间结果会存放在不同的目录，这样不会有同名文件的问题。所以你如果基于scons来编译就可以不考虑这个问题。

如果是基于MDK，MDK也不会有问题。有问题是IAR，IAR的IDE环境不允许同名文件，在IDE里好像是无解的。

pansila, 2018-06-05:
因为要兼容多个OS，照顾客户习惯，应该不会上scon，不过我试了MDK也会有冲突，因为output都在一个目录，难道哪里可以设置下？

TanekLiang, 2018-06-05:
同名文件，mkd应该是会自动重命名的。
[image]

pansila, 2018-06-05:
可能是最新的Keil功能，你是什么版本？我的v5.11.1没有这样的功能

TanekLiang, 2018-06-05:
我用的是5.25。我记得这个功能是比较早就有的。

我的操作方式是，在application目录创建一个board.c文件，这样就和driver/board.c同名了，然后使用scons生成新的工程。

pansila, 2018-06-05:
还要scons处理下project？那这看起就是scons的功能了

armink, 2018-06-05:
试试这个方法？

- 屏蔽 lwip 的 mem.c 不进行编译
- 新建 lwip_mem.c ，里面 `#include "mem.c"`
- 添加该文件进行编译

pansila, 2018-06-05:
@armink 没tool的支持只能这样了

TanekLiang, 2018-06-05:
@pansila scons和手动没有本质区别。我的意思是，我基于i.mx rt1052的bsp做了这些操作，然后在mdk里面没有出现同名问题。你可以试试能不能复现。

BernardXiong, 2018-06-08:
Keil MDK重名的问题是由scons在生成工程时自动解决的。

@pansila 
你们是哪家厂商？一些问题可以直接联系到我们来共同考虑。