RT-Thread/rt-thread #1514: 关于UART的rt_hw_serial_isr，唤醒等待线程BUG（导致fish等读取线程假死）

chinawrj, 2018-06-08:
影响版本：
  应该是所有版本。
导致的原因：
  rx_indicate唤醒的时候有可能等待线程不再等待队列中，导致等待线程永远不会被唤醒。
  ```
1152             /* invoke callback */
1153             if (serial->parent.rx_indicate != RT_NULL)
1154             {
1155                 rt_size_t rx_length;
1156
1157                 /* get rx length */
1158                 level = rt_hw_interrupt_disable();
1159                 rx_length = (rx_fifo->put_index >= rx_fifo->get_index)? (rx_fifo->put_index - rx_fifo->get_index):
1160                     (serial->config.bufsz - (rx_fifo->get_index - rx_fifo->put_index));
1161                 rt_hw_interrupt_enable(level);
1162
1163                 serial->parent.rx_indicate(&serial->parent, rx_length);
1164             }
1165             break;
1166         }
1167         case RT_SERIAL_EVENT_TX_DONE:
1168         {
"components/drivers/serial/serial.c" 1242 行 --93%--                                                    
```

具体分析：
  线程调用read函数，读取UART的数据：
```
 139 static int serial_fops_read(struct dfs_fd *fd, void *buf, size_t count)
 140 {
 141     int size = 0;
 142     rt_device_t device;
 143
 144     device = (rt_device_t)fd->data;
 145
 146     do
 147     {
 148         size = rt_device_read(device, -1,  buf, count);
 149         if (size <= 0)
 150         {
 151             if (fd->flags & O_NONBLOCK)
 152             {
 153                 size = -EAGAIN;
 154                 break;
 155             }
 156
 157             rt_wqueue_wait(&(device->wait_queue), 0, RT_WAITING_FOREVER);
 158         }
 159     }while (size <= 0);
 160
 161     return size;
 162 }
"components/drivers/serial/serial.c" 1242 行 --12%--                                                                         
```
可以看到如果没有读取到数据，并且UART的FLAGS允许NONBLOCK的话，读取UART数据的线程就会调用rt_wqueue_wait线程。这个rt_wqueue_wait源码如下：
```
 62 int rt_wqueue_wait(rt_wqueue_t *queue, int condition, int msec)
 63 {
 64     int tick;
 65     rt_thread_t tid = rt_current_thread;
 66     rt_timer_t  tmr = &(tid->thread_timer);
 67     struct rt_wqueue_node __wait;
 68
 69     tick = rt_tick_from_millisecond(msec);
 70
 71     if ((condition) || (tick == 0))
 72         return 0;
 73
 74     __wait.polling_thread = rt_thread_self();
 75     __wait.key = 0;
 76     __wait.wakeup = __wqueue_default_wake;
 77     rt_list_init(&__wait.list);
 78
 79     rt_wqueue_add(queue, &__wait);
 80
 81     /* current context checking */
 82     RT_DEBUG_NOT_IN_INTERRUPT;
 83     rt_thread_suspend(tid);
 84
 85     /* start timer */
 86     if (tick != RT_WAITING_FOREVER)
 87     {
 88         rt_timer_control(tmr,
 89                          RT_TIMER_CTRL_SET_TIME,
 90                          &tick);
 91
 92         rt_timer_start(tmr);
 93     }
 94
 95     rt_schedule();
 96
 97     rt_wqueue_remove(&__wait);
 98
 99     return 0;
100 }
101
"components/drivers/src/waitqueue.c" 101 行 --98%--         
```

可以看到这个函数是先将自己加入到wait_queue---->rt_thread_suspend 睡眠自己。

如果wait_queue-----(UART的中断调用rt_hw_serial_isr唤醒线程，线程恰好在waitqueue中，此时线程还未处于suspend状态，然后被移除)--->rt_thread_suspend 睡眠自己---->线程永远不会被唤醒（已经被移除掉了）

触发方法：

   出发这个bug的方法很假单，在putty中连接UART，在fish中，直接右击粘贴一长串字符，如果驱动采用中断接收的话，就会直接触发这个bug，fish出现不能输入，只能输出的现象。
解决方法：
  在rt_thread_suspend自己之前再次加锁保护的情况下，检查是否已经有数据ready。

BernardXiong, 2018-06-08:
好的，我们会先验证，重现下。

geniusgogo, 2018-06-11:
@chinawrj 你描述的意思是不是这样：serial_fops_read会调用rt_wqueue_wait，rt_wqueue_wait刚执行完rt_wqueue_add但是下一句rt_thread_suspend还没执行，此时来了数据就会进入serial_fops_rx_ind->rt_wqueue_wakeup.
而rt_wqueue_wakeup中会rt_wqueue_remove。所以当执行到rt_thread_suspend时，有可能wake queue已经没有rt_wqueue_add时的线程了。所以导致线程无法唤醒。

那么这样的话其实不加锁的话也可以将rt_wqueue_add(queue, &__wait);和rt_thread_suspend(tid);调换顺序

chinawrj, 2018-06-11:
@geniusgogo 调换顺序也解决不了下面这个case：
如果发生这个现象之后，UART再也不来数据，那么UART的rx_ring_buffer中还会有数据，但是再也没有UART中断触发线程唤醒了。原因很简单：最后一次唤醒等待线程时，线程没有suspend-->UART中断唤醒线程然后退出（假设以后UART没有数据，那么再也不发生UART接收中断）--->线程SUSPEND。

armink, 2018-06-11:
@chinawrj 你看下我的操作流程，可以全屏，是这样复现的吗？

[image]

我记得我前两天好像也遇到了类似问题，当时没有深入分析，最近又没察觉到了

armink, 2018-06-12:
@chinawrj 确实是，putty 容易复现

geniusgogo, 2018-06-12:
> 调换顺序也解决不了下面这个case：
如果发生这个现象之后，UART再也不来数据，那么UART的rx_ring_buffer中还会有数据，但是再也没有UART中断触发线程唤醒了。原因很简单：

最后一次唤醒等待线程时，线程没有suspend-->UART中断唤醒线程然后退出（假设以后UART没有数据，那么再也不发生UART接收中断）--->线程SUSPEND。

要彻底解决，必须要确保线程睡眠时，ring_buffer中没有数据，并且要做好race condition.

是，这样只能靠后续的数据来唤醒。最终可能就是最后的数据依然在fifo中。

chinawrj, 2018-06-12:
Hi @armink  
  感谢这么快速的相应和fix，但是这个patch似乎依然无法根本解决这个问题。

 139 static int serial_fops_read(struct dfs_fd *fd, void *buf, size_t count)
 140 {
 141     int size = 0;
 142     rt_device_t device;
 143
 144     device = (rt_device_t)fd->data;
 145
 146     do
 147     {
 148         size = rt_device_read(device, -1,  buf, count);
 149         if (size <= 0)
 150         {
 151             if (fd->flags & O_NONBLOCK)
 152             {
 153                 size = -EAGAIN;
 154                 break;
 155             }
 156             /**************唤醒发生在这里****************/
 157             rt_wqueue_wait(&(device->wait_queue), 0, RT_WAITING_FOREVER);
 158         }
 159     }while (size <= 0);
 160
 161     return size;
 162 }
"components/drivers/serial/serial.c" 1242 行 --12%--     
参考上面代码，如果UART中断唤醒开始于rt_device_read之后，结束于rt_wqueue_wait之前，那么依然有问题。其实我尝试修复过这个BUG，但是考虑要动很多东西（涉及到框架），因为需要在睡眠的那一刻（或者加锁操作的前夕）再次判断是否有数据，所以我就没有自己修复。

解决方法就两类：
1. 一是加个大的lock。BIG LOCK，锁定较长时间，锁定一开始就在serial_fops_read内判断是否有数据加锁在serial_fops_read中。
2. 二是加个小的lock，只在rt_wqueue_wait中加锁，但是需要rt_wqueue_wait回调外面的函数判断UART是否有数据。

方法1会牺牲OS的实时性，方法2稍微需要调整一下机制但是实时性较好

closed 2018-06-12 by BernardXiong

armink, 2018-06-12:
@chinawrj 

先多谢你的反馈哈

> 但是这个patch似乎依然无法根本解决这个问题。

这里问题并不是指的引出 issue “finsh等读取线程假死”的问题吧？可以先尝试用下新的 patch ，看看会不会再重新被复现出来。

你说的这个缺陷，确实还存在，但对于用户的使用可能已经不影响了吧？

chinawrj, 2018-06-12:
@armink 
的确不会假死了，只会出现需要发送方多发点数据（比如多敲一个回车），才能收完整数据的现象。

这个会影响将UART作为通信接口的应用啊。很多应用是有串口协议，格式类似于len+payload形式，然后ping+pong工作，一个request一个response，然后就会出现没有response的现象，需要再次发request才会效果。

armink, 2018-06-12:
@chinawrj 
好的，现在至少用户体验上不会假死了。

我再来看下怎么解决，这个比较极端情况下的问题。

armink, 2018-06-12:
@chinawrj 我这里复现这种极端情况需要一定概率。而且发现蛮奇怪的，总是粘贴到 putty 的 **最后一个字节** 会出现跟下一包数据粘连的情况，粘贴的中间数据不会出现。

armink, 2018-06-12:
```
            +----------------------+------------------------+
            |    finsh_thread      |        ISR             |
            +-----------------------------------------------+
            |                      |                        |
            |                      |                        |
            |    getchar()         |                        |
            |        +             |                        |
            |        |             |                        |
            |        v             |                        |
            |     _read_r()        |                        |
            |        +             |                        |
            |        |             |                        |
            |        v             |                        |
            |      read()          |                        |
            |        +             |                        |
            |        |             |                        |
            |        v             |                        |
            |   dfs_file_read()    |                        |
            |        +             |                        |
            |        |             |                        |
            |        v             |                        |
+----+------+  serial_fops_read()  |                        |
     ^      |        +             |                        |
     |      |        |             |  USART1_IRQHandler()   +----+-----+
     |      |        |             |          +             |    ^
     |      |        |             |          |             |    |
     |      |        |             |          v             |    |
     |      |        |             |    rt_hw_serial_isr()  |    |
     |      |        |             |          +             |    +
     +      |        |             |          |             |    T2
     T1     |        |             |          v             |    +
     +      |        |             | serial_fops_rx_ind()   |    |
     |      |        |             |          +             |    |
     |      |        |             |          |             |    |
     |      |        |             |          v             |    v
     |      |        |             |   rt_wqueue_wakeup()   +----------+
     |      |        |             |                        |
     |      |        |             |                        |
     v      |        v             |                        |
+-----------+    rt_wqueue_wait()  |                        |
            |                      |                        |
            +----------------------+------------------------+
```

如上图，感觉问题的本质应该是 finsh getchar 时，`serial_fops_read` 里检测到串口 **没数据** ，到 `rt_wqueue_wait` 里真正把finsh 线程挂起的这段时间 **T1** 里，ISR 有足够机会提前执行唤醒 finsh 的动作。

解决这个问题，感觉有以下几个思路：

-  **【思路1】T1 阶段完全关闭中断** ：这个阶段牵扯到蛮多函数调用，有些阶段强制关闭中断的面积又不能太大，所以导致 T1 阶段肯定会有开中断的瞬间，此时 ISR 及可能会立刻执行，前面工作就白做了。所以这条路好像不太好走通
-  **【思路2】推迟执行唤醒 finsh 的代码** ：可否尝试在中断中不要直接 `rt_wqueue_wakeup` 代码，把这部分工作交给其他线程去完成，错过 T1 时间段。这样即便在 T1 阶段触发中断，ISR 的 T2 时间里也不会真正执行 `rt_wqueue_wakeup` 
- **【思路3】改进线程挂起机制** ：在线程执行真正的挂起动作时（目前应该是在 `rt_schedule` 里 ），如果之前有其他地方对其执行了恢复操作，线程可否不用继续挂起，而是直接执行。不知道这点可行性有多大。

我的考虑不一定周到， @BernardXiong @chinawrj @geniusgogo @aozima 大家也多提提意见吧

chinawrj, 2018-06-12:
@armink 图差不多描述的是对的。其实只要在线程真正suspend前一刻，能够检查一下ringbuffer是否又有数据就OK了。

geniusgogo, 2018-06-15:
> 其实只要在线程真正suspend前一刻，能够检查一下ringbuffer是否又有数据就OK了。

这不是原子操作，所以实现不了的。
你检查完了还是要去suspend，那么在检查完后，suspend前还是会来数据。
其实只要在rt_wqueue_wait(&(device->wait_queue), 0, RT_WAITING_FOREVER);改成rt_wqueue_wait(&(device->wait_queue), 0, 1);这个1可以改成计算传输速率换算成tick。或者重构成一个具备守护进程的模式。只要fifo有数据那么就通过守护进程去唤醒wait队列。

为什么这个issue关闭了？

chinawrj, 2018-06-15:
@geniusgogo 很明显检查ringbuffer是否有数据需要关闭中断。

geniusgogo, 2018-06-15:
1.检查ringbuffer是需要关闭中断。我说的你不可能做到原子操作。所以你在从readd到suspend不可能一直关闭中断。
2.改成非RT_WAITING_FOREVER，只会增加CPU占用。其他没有什么吧？这又没有什么标准,影响不大。你要知道1个tick的等待超时这是某某OS的tick调度策略。所以考虑CPU负载太过保守了。
3.那就重构waitqueue。已经和armink讨论过，应该会这么改。

chinawrj, 2018-06-15:
@geniusgogo 
UART提供一个回调函数，在调用rt_wqueue_wait的时候。然后在rt_wqueue_wait中，rt_wqueue_wait的代码在真正suspend之前，关闭中断调用UART提供的回调函数，这个回调函数告诉rt_wqueue_wait是否还要继续睡眠（其实就是ringbuffer有没有数据）。问题解决

这个bug的解法需要改一下waitqueue，不然我早自己修复了，因为涉及到已有框架的问题。
另外关于你的RT_WAITING_FOREVER改法，work around是CPU占用是其次，最大的问题延时！延时！延时！说3遍，导致偶发的数据接收延时！
 
不用咬文嚼字了，
解决方法：
1. 睡眠前检查ringbuffer
2. 改RT_WAITING_FOREVER不靠谱

chinawrj, 2018-06-17:
@armink 
可以不用回调函数，只检查flag的，这个flag放在waitqueue相关的结构体中，有数据的时候就置位，没数据就不置位。置位的时候，由UART ISR负责。waitqueue中负责关中断访问这个flag，并且在关中断的情况下搞定suspend之类的。
思路就是这样的，就是waitqueue中进行最后确认是否睡眠。至于代码框架设计问题，就是可以商讨的很多了，比如我之前说的回调函数（稍微重量级一点），或者flag（这个方法应该还行，而且没有框架上也没问题）

> 这样实现好像不太难，你有没有测试过？ 我感觉结果会是这个数据来的时候，不通知 finsh 线程了，最终还是会少唤醒一次 finsh 线程，除非下次又有数据过来。

没有测试过，但是我考虑过在逻辑上是没有少唤醒问题的，除非实现上出问题了，或者你设想和实现方式和我的不一样。如果你觉得有问题，可以画出来，我们讨论一下。

我觉得这个是wait_queue设计问题，最好放在waitqueue相关的代码里面一并解决。

geniusgogo, 2018-06-20:
@chinawrj 内核关中断,调用用户外部的回调，你觉得会有这样的内核设计方式吗？
另外

> 在真正suspend之前，关闭中断调用UART提供的回调函数
其实就是将read->suspend一直关中断的方式缩小了关中断的区间。本质上还是关中断检查是否有数据，read本身你可以理解为一种检查是否有数据。所以效果等效于read->suspend一直关中断。只是你说的这方式在缩小关中断的区间。

> 最大的问题延时！延时！延时！重要事情说3遍。会导致偶发的数据接收延时，延时有多大呢？取决于你设置的时长！

我也重要的事情说三遍，wait 1个tick相当于延时了一个os tick，那么一个os tick里绝对能保证得到调度。所以我强调了在shell上说延时 是保守的想法。当然这样的实现肯定不是最优的解决方式。

> 我从来没说过要一直关闭中断，我的意思一直都很简单：检查ringbuffer就行了。只需要检查的时候关闭就OK。

上面已经说了在真正suspend前关闭了中断调用用户回调显然就是扯淡的设计。还不如从read到suspend一直关中断，suspend后再开。避免内核被煞笔煽耳光。

geniusgogo, 2018-06-20:
@chinawrj 首先对于shell来说不存在你说的 延时 延时 延时 这种场景。1个os tick的wait对你不是机器人变身的正常人来说不存在使用shell时的延时感觉。另外这种设计肯定不是最优解，这个肯定得承认，并且根本就不会这样去实现。只是可以这样最小化的去改动。作为一种临时性的最小改动方案。

另外在正常情况下我说的不可能从read—>suspend一直关中断，意思是read和suspend必须是原子操作。那么你说在真正的suspend时关中断掉用回调检查，这和read->suspend原子操作不是一个实现思路吗？你这样做的优点仅仅是相对于来说缩小了关闭中断的区间。但是本质来说还是一样的思路。再说明一下，为了解决这个问题，莫名其妙的在suspend前调一个回调，这是干嘛呢？换个人，换个场景看OS的代码时他懵不懵逼？另外通常的OS组件代码在回调用户接口的时候少之又少何况是关了中断再回调？迟早被哪个煞笔打脸。

然后总结一下，我提供过两个解决方案，一个就是被你吐槽的将wait tick 改成1个tick的方案，另一个就是我说的做个守护进程的方式，也就是@armink说的第三种方式。但凡有点设计经验的人都知道我说的第一种方式不是最终解决方案，只是一种站在现在的代码实现方式下的一种最小改动方案。投机取巧罢了。
那么针对第二种方案，你为啥没发表什么意见？逮住别人的短板大声宣扬，生怕其他人看不到？看到别人的有点选择性的忽略是吧？讨论问题不用太尖锐的词语来抨击谁，这样在社区混不下去的。

BernardXiong, 2018-06-20:
@geniusgogo 莫急，我找元杰来看看。这个问题我有些概念（还没完全建立起概念），期望有时间的时候上板子上试试。应该比较好的处理方式是，在最后冻结上下文（关闭中断，或关闭调度器等）前，再去看看是否有遗留数据的情况。

geniusgogo, 2018-06-20:
@BernardXiong 我不急呢，社区原本就是来积极发言的，只是希望不要携带个人情绪。

另外问一下，为啥一定要用waitqueue，比如用event/sem不行？

geniusgogo, 2018-06-20:
`From 7dbfe1ccd6ceb61a867e1f1a791bf071b03de53a Mon Sep 17 00:00:00 2001
From: xieyangrun <xieyangrun@smartcomma.com>
Date: Wed, 20 Jun 2018 15:05:22 +0800
Subject: [PATCH] fixed wait queue

---
 components/drivers/include/ipc/waitqueue.h |  8 +++++++-
 components/drivers/src/waitqueue.c         | 24 ++++++++++++++++++----
 2 files changed, 27 insertions(+), 5 deletions(-)

diff --git a/components/drivers/include/ipc/waitqueue.h b/components/drivers/include/ipc/waitqueue.h
index 1746d0f1c..c6910ac31 100644
--- a/components/drivers/include/ipc/waitqueue.h
+++ b/components/drivers/include/ipc/waitqueue.h
@@ -5,7 +5,13 @@
 
 struct rt_wqueue_node;
 
-typedef rt_list_t rt_wqueue_t;
+struct rt_wqueue
+{
+    rt_list_t list;
+    rt_uint32_t wake_counter;
+};
+typedef struct rt_wqueue rt_wqueue_t;
+
 typedef int (*rt_wqueue_func_t)(struct rt_wqueue_node *wait, void *key);
 
 struct rt_wqueue_node
diff --git a/components/drivers/src/waitqueue.c b/components/drivers/src/waitqueue.c
index 7fff2d382..3a88f9881 100644
--- a/components/drivers/src/waitqueue.c
+++ b/components/drivers/src/waitqueue.c
@@ -36,11 +36,16 @@ void rt_wqueue_wakeup(rt_wqueue_t *queue, void *key)
 
     struct rt_list_node *node;
     struct rt_wqueue_node *entry;
-
+    
+    level = rt_hw_interrupt_disable();
+    
     if (rt_list_isempty(queue))
+    {
+        queue->wake_counter++;
+        rt_hw_interrupt_enable(level);
         return;
+    }
 
-    level = rt_hw_interrupt_disable();
     for (node = queue->next; node != queue; node = node->next)
     {
         entry = rt_list_entry(node, struct rt_wqueue_node, list);
@@ -74,13 +79,24 @@ int rt_wqueue_wait(rt_wqueue_t *queue, int condition, int msec)
 
     if ((condition) || (tick == 0))
         return 0;
-
+    
     __wait.polling_thread = rt_thread_self();
     __wait.key = 0;
     __wait.wakeup = __wqueue_default_wake;
     rt_list_init(&__wait.list);
-
+    
+    /* disable interrupt */
     level = rt_hw_interrupt_disable();
+
+    if (queue->wake_counter > 0)
+    {
+        queue->wake_counter--;
+
+        /* enable interrupt */
+        rt_hw_interrupt_enable(level);
+        return 0;
+    }
+
     rt_wqueue_add(queue, &__wait);
     rt_thread_suspend(tid);
 
-- 
2.17.1.windows.2

`

geniusgogo, 2018-06-20:
[https://github.com/geniusgogo/rt-thread/commit/7dbfe1ccd6ceb61a867e1f1a791bf071b03de53a](url)
@BernardXiong @armink 来看一下我这样实现是否可以。

chinawrj, 2018-06-24:
@geniusgogo 对事不对人。非常抱歉！如果有些用词引起了你的误会。
其实平时工作中，我们也会为有些技术问题争得面红耳赤，但是也仅仅是讨论技术问题。不会延伸到针对个人。

BernardXiong, 2018-06-25:
@geniusgogo 的PR貌似不错了的，我也改了一版本，最终改出来的情况和你的差不多，今天我会先提供出来，一些细节情况我再想下。

这个问题我昨天用QEMU跑下来，基本情况是，
read任务read后返回空数据，然后要去挂起；在要挂起而没挂起之前（read返回空数据之后）。有数据达到，然后回调了rx_ind。rx_ind中会去检查wait queue，发现里面没有挂起的任务，然后直接返回了。

所以是，1. 既没有唤醒任务；2. read任务也在这个空隙之后，就挂起了。

然后数据就遗留在其中，而read任务依然还是挂起状态。

我的解法和 @geniusgogo  差不多，嗯， @chinawrj 也提及到了：

对wqueue结构体做些调整，增加flag字段。当做wqueue_wakeup时，这个flag无论如何会置位的。

PR #1528 (2018-06-12): [components][drivers] Add irq lock to rt_wqueue_wait.
fixed #1514