RT-Thread/rt-thread #1527: vscode工程生成includepath有问题

geniusgogo, 2018-06-12:
"includePath": [
	"\"E:/project_code/lite/applications\",",
	"\"E:/project_code/lite/asflib_config\",",
	"\"E:/project_code/lite/asflib_config/clock_samd20\",",
	"\"E:/project_code/lite/board\",",
	"\"E:/project_code/lite/sam_d2x_asflib/CMSIS/Include\",",
	"\"E:/project_code/lite/sam_d2x_asflib/common/boards\",",
	"\"E:/project_code/lite/sam_d2x_asflib/common/utils\",",
	"\"E:/project_code/lite/sam_d2x_asflib/common/utils/interrupt\",",
	"\"E:/project_code/lite/sam_d2x_asflib/common2/services/delay\",",
	"\"E:/project_code/lite/sam_d2x_asflib/common2/services/delay/sam0\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/adc\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/adc/adc_sam_d_r_h\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/extint\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/extint/extint_sam_d_r_h\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/nvm\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/port\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/rtc\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/sercom\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/sercom/i2c\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/sercom/usart\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/clock\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/clock/clock_samd20\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/interrupt\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/interrupt/system_interrupt_samd20\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/pinmux\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/power/power_sam_d_r_h\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/reset/reset_sam_d_r_h\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/wdt\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/cmsis/samd20/include\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/cmsis/samd20/source\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/header_files\",",
	"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/preprocessor\",",
	**"E:/project_code/lite/mdk",
	"E:/project_code/lite/middleware/rt-thread/components/dfs/include",
	"E:/project_code/lite/middleware/rt-thread/components/drivers/include",
	"E:/project_code/lite/middleware/rt-thread/components/drivers/spi",
	"E:/project_code/lite/middleware/rt-thread/components/libc/compilers/minilibc",
	"E:/project_code/lite/middleware/rt-thread/include",
	"E:/project_code/lite/middleware/rt-thread/libcpu/arm/common",
	"E:/project_code/lite/middleware/rt-thread/libcpu/arm/cortex-m0"**
],
"defines": [
	"DEBUG",
	"EXTINT_CALLBACK_MODE=true",
	"HAVE_CCONFIG_H",
	"I2C_MASTER_CALLBACK_MODE=false",
	"RTC_COUNT_ASYNC=true",
	"RT_USING_MINILIBC",
	"USART_CALLBACK_MODE=false",
	"__SAMD20G18__"
],
"browse": {
	"path": [
		"\"E:/project_code/lite/applications\",",
		"\"E:/project_code/lite/asflib_config\",",
		"\"E:/project_code/lite/asflib_config/clock_samd20\",",
		"\"E:/project_code/lite/board\",",
		"\"E:/project_code/lite/sam_d2x_asflib/CMSIS/Include\",",
		"\"E:/project_code/lite/sam_d2x_asflib/common/boards\",",
		"\"E:/project_code/lite/sam_d2x_asflib/common/utils\",",
		"\"E:/project_code/lite/sam_d2x_asflib/common/utils/interrupt\",",
		"\"E:/project_code/lite/sam_d2x_asflib/common2/services/delay\",",
		"\"E:/project_code/lite/sam_d2x_asflib/common2/services/delay/sam0\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/adc\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/adc/adc_sam_d_r_h\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/extint\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/extint/extint_sam_d_r_h\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/nvm\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/port\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/rtc\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/sercom\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/sercom/i2c\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/sercom/usart\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/clock\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/clock/clock_samd20\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/interrupt\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/interrupt/system_interrupt_samd20\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/pinmux\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/power/power_sam_d_r_h\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/system/reset/reset_sam_d_r_h\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/drivers/wdt\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/cmsis/samd20/include\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/cmsis/samd20/source\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/header_files\",",
		"\"E:/project_code/lite/sam_d2x_asflib/sam0/utils/preprocessor\",",
		"E:/project_code/lite/mdk",
		"E:/project_code/lite/middleware/rt-thread/components/dfs/include",
		"E:/project_code/lite/middleware/rt-thread/components/drivers/include",
		"E:/project_code/lite/middleware/rt-thread/components/drivers/spi",
		"E:/project_code/lite/middleware/rt-thread/components/libc/compilers/minilibc",
		"E:/project_code/lite/middleware/rt-thread/include",
		"E:/project_code/lite/middleware/rt-thread/libcpu/arm/common",
		"E:/project_code/lite/middleware/rt-thread/libcpu/arm/cortex-m0"

geniusgogo, 2018-06-12:
加了黑体的是正常的路径，其他的路径为什么多了双引号？

BernardXiong, 2018-06-12:
这样子啊，好的，我修下。也欢迎提PR哈

geniusgogo, 2018-06-13:
`def GenerateCFiles(env):
    """
    Generate c_cpp_properties files
    """
    if not os.path.exists('.vscode'):
        os.mkdir('.vscode')
    print("\n")
    vsc_file = file('.vscode/c_cpp_properties.json', 'wb')
    if vsc_file:
        info = utils.ProjectInfo(env)

        cc = os.path.join(rtconfig.EXEC_PATH, rtconfig.CC)
        cc = os.path.abspath(cc).replace('\\', '/')

        config_obj = {}
        config_obj['name'] = 'Win32'
        config_obj['defines'] = info['CPPDEFINES']
        config_obj['intelliSenseMode'] = 'clang-x64'
        config_obj['compilerPath'] = cc
        config_obj['cStandard'] = "c99"
        config_obj['cppStandard'] = "c++11"

        # format "a/b," to a/b. remove first quotation mark("),and remove end (",)
        includePath = []
        for i in info['CPPPATH']:
            if i[0] == '\"' and i[len(i) - 2:len(i)] == '\",':
                includePath.append(i[1:len(i) - 2])
            else:
                includePath.append(i)
        config_obj['includePath'] = includePath

        json_obj = {}
        json_obj['configurations'] = [config_obj]

        vsc_file.write(json.dumps(json_obj))
        vsc_file.close()

    return`

这么改可以？

closed 2018-06-14 by geniusgogo