RT-Thread/rt-thread #1548: rt_components_init()

flightxgf, 2018-06-18:
https://github.com/RT-Thread/rt-thread/blob/5efad2a2b86797af7427ab5ee1e7e5f06b6b45b4/bsp/stm32f429-apollo/applications/application.c#L53

application.c中rt_init_thread_entry线程初始化rt_components_init(); 和 在board.c中初始化rt_components_board_init(); 总共初始化两次 是为何？

TanekLiang, 2018-06-18:
一个是在调度器起来之前调用的，一个是在调度器起来之后调用的。

有些组件，例如串口，例如sdram的初始化，是放在调度器起来之前最好。
有些组件，例如sdio部分，gui部分，可以放在调度器起来之后做。

所以有了这两个区分。

flightxgf, 2018-06-18:
---有些组件，例如串口，例如sdram的初始化，是放在调度器起来之前最好-那board.c中应该把rt_console_set_device(RT_CONSOLE_DEVICE_NAME);放在rt_components_board_init后面吧

TanekLiang, 2018-06-18:
只要求是在uart初始化之后，如果是自动初始化，放在rt_components_board_init后面也很合适的。

有时候为了调试在rt_components_board_init的组件，会把uart提前初始化并且设置rt_console_set_device。这样就可以在具体驱动里面输出log了 。

flightxgf, 2018-06-18:
了解，这样的操作就会出现两次初始化，我建议hw_usart_init初始化不要设置INIT_BOARD_EXPORT，

flightxgf, 2018-06-18:
从目前的bsp来看hw_usart_init初始化执行了三次 ，初始化一次就购了吧

TanekLiang, 2018-06-18:
不是，只应该调用一次。

如果手动了，就不要加入自动初始化。
如果是自动初始化，那么取决于类型，BOARD的才在rt_components_board_init里面调用。
其他的在rt_components_init里调用。

flightxgf, 2018-06-18:
ok，s是的，手动是一次，目前的bsp是自动的，而且也有手动，执行了两次hw_usart_init，所以我建议不要自动uart，直接手动这样就执行一次hw_usart_init，然后rt_console_set_device在rt_components_board_init前面就是合理的

TanekLiang, 2018-06-18:
那是我的疏忽了，能帮忙提交pr来修复吗？

flightxgf, 2018-06-19:
可以

mysterywolf, 2021-02-28:
#1551

closed 2021-02-28 by mysterywolf