RT-Thread/rt-thread #1553: rt_components_init初始化

flightxgf, 2018-06-20:
https://github.com/RT-Thread/rt-thread/blob/5efad2a2b86797af7427ab5ee1e7e5f06b6b45b4/bsp/stm32f429-apollo/applications/application.c#L53

1. 测试发现去掉rt_components_init不会进入shell比如msh，这里只是初始化一些资源，为何会导致跟msh关联

2. uart1使用shell，uart2使用通信，去掉rt_components_init发现uart2通信正常，加入rt_components_init发现可以进入shell，**但是uart也会有正常通信的数据，但是会有很多垃圾数据出现**
垃圾数据如下：
"`鎜骟鰃出w甾g出o出gf篥w出o出篥gf篥篥s出篥o出gf篥gf辸出篥o出wf篥gf篥gf篥gf辡出o出辡出o出篥gf篥gf篥s出?`"

查看源码经分析还是没找到rt_components_init加入为何uart2会出现如上的垃圾数据

TanekLiang, 2018-06-21:
你可以在rtconfig.h 把RT_DEBUG_INIT 的值定义为1， 这样就可以看到 rt_components_init的具体工作。
这里去创建shell的线程的。

flightxgf, 2018-06-21:
哈哈哈，看来呜呜呜呜呜是法宝
我试过了，rt_components_init的具体工作是：有eth初始化，I2C初始化，SD卡，还有加载shell，
我的疑惑是 注释掉rt_components_init ，uart2 通信ok，放开rt_components_init，uart2会有通信数据，同时会打印出很多不认识的垃圾数据
备注：uart1用作shell，uart2使用通信

TanekLiang, 2018-06-21:
可能是驱动部分有问题，导致不能两个uart同时使用。

flightxgf, 2018-06-21:
谢谢谢谢，我再检查下。估计驱动得做些改动，来区分uart1和uart2，我看引脚ok，我是直接使用了bsp中的drv_uart.c

flightxgf, 2018-06-22:
建议官方 把此bsp  uart2打开，测试看看rt_components_init  存在的情况下为何uart2会出现很多垃圾数据（
备注： 裸机测试uart1 ,uart2 共存时，数据都ok，在 此bsp中测试时单独测试各自ok，共存时就会有乱码（已检查各自驱动），我准备使用nano测试下看看，如果nano下ok，就说明rt_components_init  中确实发生了什么

mysterywolf, 2021-02-28:
请问该问题解决了吗？

mysterywolf, 2021-02-28:
#1551

closed 2021-02-28 by mysterywolf