RT-Thread/rt-thread #1569: RTT_EXE_PATH错误

Faynman, 2018-06-25:
安装环境:
VirtualBox虚拟机上安装Ubuntu18系统
已经安装好了scons
rt-thread源码win10上的git下载好的,然后拷贝到ubuntu上(因为ubuntu上git克隆不了rt-thread源码, 可能是网络问题?)

错误信息:
@VirtualBox:~$ cd rt-thread/bsp/allwinner_tina/
@VirtualBox:~/rt-thread/bsp/allwinner_tina$ scons --menuconfig
scons: Reading SConscript files ...
KeyError: 'RTT_EXEC_PATH':
  File "/home/li/rt-thread/bsp/allwinner_tina/SConstruct", line 24:
    objs = PrepareBuilding(env, RTT_ROOT, has_libcpu=True)
  File "../../tools/building.py", line 252:
    if os.environ['RTT_EXEC_PATH']:
  File "/usr/lib/python2.7/UserDict.py", line 40:
    raise KeyError(key)
@VirtualBox:~/rt-thread/bsp/allwinner_tina$ 

Google了也没找到解决方法,只能来这里发个issuse, 有人知道怎么回事吗?

BernardXiong, 2018-06-25:
不行哦，你win10下clone下来，这些分行都是windows分行符，然后再丢到linux下铁定就不行了。

需要linux下独立去git clone

Faynman, 2018-06-25:
@BernardXiong 
在ubuntu16环境下，git clone测试还是一样的结果。

BernardXiong, 2018-06-26:
好的，我找linux环境来试试

BernardXiong, 2018-06-26:
我用这个环境并没有问题呢，
ubuntu@VM-8-33-ubuntu:~/workspace/rt-thread/bsp/allwinner_tina$ uname --all
Linux VM-8-33-ubuntu 4.4.0-91-generic #114-Ubuntu SMP Tue Aug 8 11:56:56 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

你的配置中是否没配置工具链目录，或者RTT_EXEC_PATH环境变量设置有错？

必须的，
要么RTT_EXEC_PATH环境变量设置没问题；
要么rtconfig.py中工具链路径（EXEC_PATH）没问题；

Faynman, 2018-06-26:
RTT_EXEC_PATH和工具链目录不懂怎么设置,请问有教程吗?
我也奇怪,看这个教程,并没有教设置环境变量之类的,但是git克隆的会出现问题,下载released版本就能够通过测试.

Faynman, 2018-06-26:
@BernardXiong

Faynman, 2018-06-26:
添加环境变量可以解决了
我的arm-none-eabi-gcc默认安装在/usr/bin

cd ~
sudo nano .bashrc
#在最后行添加环境变量
export RTT_EXEC_PATH="/usr/bin/arm-none-eabi-gcc"
#更新环境变量
source .bashrc
#查看环境变量
export

BernardXiong, 2018-06-27:
`if 'RTT_EXEC_PATH' in os.environ:`

这样也没效果？或者说，rtconfig.py中设置正确的EXEC_PATH也没效果？

Faynman, 2018-06-29:
修改rtconfig.py的方法我还没测试, 因此不知道可不可以.

在rtconfig.py脚本里,这个if应该是判断是否存在"RTT_EXEC_PATH"环境变量.
```
if os.getenv('RTT_EXEC_PATH'):
    EXEC_PATH = os.getenv('RTT_EXEC_PATH')
```
实际上我的环境变量里是没有"RTT_EXEC_PATH"的,因此会报错.
建议修改下说明文档,让大家记得添加环境变量.

closed 2021-11-17 by mysterywolf