RT-Thread/rt-thread #1594: micropython found fatal error: dfs_posix.h

Afantor, 2018-06-30:
In win10, when I add micropython to build the project, I found some error.
on env logs:

CC build\packages\micropython-latest\port\mpy_main.o 
CC build\packages\micropython-latest\port\rtt_getchar.o 
CC build\packages\micropython-latest\port\_frozen_mpy.o 
packages\micropython-latest\port\mpy_main.c:32:23: fatal error: dfs_posix.h: No such file or directory compilation terminated. 
scons: *** [build\packages\micropython-latest\port\mpy_main.o] Error 1 
scons: building terminated because of errors.

Now,How I can do it?

TanekLiang, 2018-06-30:
@armink

SummerGGift, 2018-07-01:
在env 里面开启 支持 dfs 文件系统，并开启 POSIX API 支持即可。

TanekLiang, 2018-07-01:
@SummerGGift 不会自动添加这个依赖吗？

BernardXiong, 2018-07-01:
@SummerGGift 是的，能否做得更人性化一些呢？

Afantor, 2018-07-01:
@SummerGGift 能不能提供详细的配置步骤？microPython开发手册里没有详细的menuconfig配置说明。
当我在env里面开启dfs文件系统和开启POSIX API后，重新scons出现新的错误：
CC build/packages/micropython/port/machine_pin.o
CC build/packages/micropython/port/machine_uart.o
In file included from packages/micropython/py/mpstate.h:35:0,
from packages/micropython/py/runtime.h:29,
from packages/micropython/port/machine_uart.c:30:
packages/micropython/port/machine_uart.c:216:29: error: static declaration of 'machine_uart_locals_dict' follows non-static declaration
STATIC MP_DEFINE_CONST_DICT(machine_uart_locals_dict, machine_uart_locals_dict_table);
^
packages/micropython/py/obj.h:309:25: note: in definition of macro 'MP_DEFINE_CONST_DICT'
const mp_obj_dict_t dict_name = { 
^
In file included from packages/micropython/port/machine_uart.c:36:0:
packages/micropython/port/machine_uart.h:34:28: note: previous declaration of 'machine_uart_locals_dict' was here
extern const mp_obj_dict_t machine_uart_locals_dict;
^
packages/micropython/port/machine_uart.c: In function 'machine_uart_ioctl':
packages/micropython/port/machine_uart.c:235:5: warning: return makes integer from pointer without a cast
return NULL;
^
scons: *** [build/packages/micropython/port/machine_uart.o] Error 1
scons: building terminated because of errors.

gcc版本是使用env自带的5.4.1版本。

BernardXiong, 2018-07-01:
是哪个bsp呢？RTT是github版本 + package上的mpy？

Afantor, 2018-07-01:
@BernardXiong 是stm32f10x-HAL的

BernardXiong, 2018-07-02:
So, @SummerGGift try it please.

SummerGGift, 2018-07-02:
@Afantor 
MicroPython 默认不需要开启 dfs，所以不需要默认开启 dfs 相关的选项，我看你的错误是缺少 dfs 相关的头文件就让你打开那个宏。已经对这个问题提了 PR，不会再默认包含 dfs 的头文件。

另外我看到你的编译错误中还有已经修复过的串口方面的错误提示，说明你的现在的软件包不是最新最新版，推荐更新一下软件包。

stm32f10x-HAL 我在 mdk 和 gcc 下都重新测试了一下，可以编译通过，需要注意的是要在MDK中选择一个资源足够的芯片，在　gcc　的链接脚本也要改大默认的芯片大小。在 stm32_rom_ld 文件中修改 LENGTH 选项。

/* Program Entry, set to mark it as "used" and avoid gc */
MEMORY
{
    CODE (rx) : ORIGIN = 0x08000000, LENGTH = 512k /* 512KB flash */
    DATA (rw) : ORIGIN = 0x20000000, LENGTH =  64k /* 64K sram */
}

closed 2018-07-05 by BernardXiong