RT-Thread/rt-thread #1607: 工具链 Python3支持依然有问题

airystar, 2018-07-05:
> pkgs --upgrade
Traceback (most recent call last):
  File "~/.env/tools/scripts/env.py", line 31, in <module>
    from cmds import *
  File "~/.env/tools/scripts/cmds/cmd_package.py", line 112
    except Exception, e:
                    ^
SyntaxError: invalid syntax

> python --version
Python 3.6.5

BernardXiong, 2018-07-05:
好的，我们升级上去看看这个问题

aozima, 2018-09-03:
PING

pcdack, 2018-09-20:
可以将环境强制弄成`python2`。将`pkgs`脚本中的`python`改为`python2`

aozima, 2019-01-14:
这个问题应该在 #1952 修复了

ziyouchutuwenwu, 2019-09-25:
今天看，还没修复，python3还没支持，
```
rm -rf ~/.env
scons --menuconfig
```
以后，~/.env/tools/scripts/pkgs，一样报错

SummerGGift, 2019-09-25:
> 今天看，还没修复，python3还没支持，
> 
> ```
> rm -rf ~/.env
> scons --menuconfig
> ```
> 
> 重新git clone 到env，~/.env/tools/scripts/pkgs，一样报错

可以参考下面这篇文章便捷的切换 Python 版本：
https://blog.csdn.net/beijiu5854/article/details/77897767

ziyouchutuwenwu, 2019-09-25:
> > 今天看，还没修复，python3还没支持，
> > ```
> > rm -rf ~/.env
> > scons --menuconfig
> > ```
> > 
> > 
> > 重新git clone 到env，~/.env/tools/scripts/pkgs，一样报错
> 
> 可以参考下面这篇文章便捷的切换 Python 版本：
> https://blog.csdn.net/beijiu5854/article/details/77897767

切换py版本知道，也感谢你的回复，只是现在py3是主流了，却只支持py2, 有点不太理解，希望能尽快换成py3

mysterywolf, 2021-03-14:
挖坟来了，现在还有少数sconscript采用python2语法 最近莫名其妙开始报错了https://github.com/RT-Thread/rt-thread/pull/4434

mysterywolf, 2021-03-16:
https://github.com/RT-Thread/rt-thread/pull/4466

closed 2021-03-20 by mysterywolf