RT-Thread/rt-thread #1654: Convert License from GPLv2+ to Apache License v2

BernardXiong, 2018-07-22:
考虑把RT-Thread主许可证从GPLv2+更改为Apache License v2许可证，主体覆盖include/src，components，libcpu等文件夹下的代码。

而对于BSP目录，会尽量转换，但不包括厂商提供的固件库等。

liruncong, 2018-07-23:
您好,能问下这两个license有什么区别吗?改用Apache License v2
后,是否能保证RT-thread操作系统外的应用程序代码(比如客户自己领域的专业应用)安全?

BernardXiong, 2018-07-24:
代码安全不是由许可证保证的啊

关于Apache License v2的文本可以参考以下链接：

* [Apache License v2](https://www.apache.org/licenses/LICENSE-2.0.html)
* [Apache License v2中文版本](http://www.jinbuguo.com/apache/menu22/license.html)
* [Apache许可证](https://zh.wikipedia.org/wiki/Apache%E8%AE%B8%E5%8F%AF%E8%AF%81)

BernardXiong, 2018-07-24:
一些常见许可协议情况，也可以参看这份文章

* [一张图看懂开源许可协议](https://blog.csdn.net/testcs_dn/article/details/38496107)

RT-Thread现有（<v3.0.4）许可证协议是一种GPLv2+，即GPLv2许可的修改版本，有些类似于LGPLv2.1，但是因为在嵌入式领域并不像PC上，可以以library方式发布，所以现有选择的是一种应用于嵌入式领域的类似LGPLv2.1许可的GPLv2+版本。

liruncong, 2018-07-24:
最后这张图看明白了.前面那几个许可证,老实说,看得雨里雾里,也没耐心看下去.谢谢@BernarXiong解惑!

mbbill, 2018-09-18:
emm... 这样的重磅好消息竟然没有人关注

closed 2019-07-21 by BernardXiong