RT-Thread/rt-thread #1674: 在beagleBone Black rev3版本上运行死在任务切换

zipper1956, 2018-07-29:
2.X和3.0.x的版本中，在beagleBone Black rev3上运行死在任务切换上，使用串口调试之后发现是死在任务切换的最后一句恢复寄存器上，即“ldmfd sp!, {r0-r12, lr, pc}^)”，请问各位有遇到的吗？谢谢！

BernardXiong, 2018-07-29:
这个好像不应该吧，你确认下是否是编译器的问题

zipper1956, 2018-07-30:
感谢回复，之前也试过mentor的其他版本编译器，都会遇到这个问题，反汇编elf文件中“ldmfd sp!, {r0-r12, lr, pc}^)”这句代码展开也没看出什么问题，请问您用的什么版本的编译器？我目前用的版本如下。
scons版本：2.2
python版本：2.7
编译器：Sourcery_CodeBench_Lite_for_ARM_EABI，版本信息如下
C:\Sourcery_CodeBench_Lite_for_ARM_EABI\bin>arm-none-eabi-gcc.exe -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc.exe
Target: arm-none-eabi
Configured with: /scratch/maciej/arm-eabi-2014.05-rel/src/gcc-4.8-2014.05/config
ure --build=i686-pc-linux-gnu --host=i686-mingw32 --target=arm-none-eabi --enabl
e-threads --disable-libmudflap --disable-libssp --disable-libstdcxx-pch --enable
-extra-sgxxlite-multilibs --with-gnu-as --with-gnu-ld --with-specs='%{save-temps
: -fverbose-asm} -D__CS_SOURCERYGXX_MAJ__=2014 -D__CS_SOURCERYGXX_MIN__=5 -D__CS
_SOURCERYGXX_REV__=28' --enable-languages=c,c++ --disable-shared --enable-lto --
with-newlib --with-pkgversion='Sourcery CodeBench Lite 2014.05-28' --with-bugurl
=https://sourcery.mentor.com/GNUToolchain/ --disable-nls --prefix=/opt/codesourc
ery --with-headers=yes --with-sysroot=/opt/codesourcery/arm-none-eabi --with-bui
ld-sysroot=/scratch/maciej/arm-eabi-2014.05-rel/install/host-i686-mingw32/opt/co
desourcery/arm-none-eabi --with-libiconv-prefix=/scratch/maciej/arm-eabi-2014.05
-rel/obj/pkg-2014.05-28-arm-none-eabi/arm-2014.05-28-arm-none-eabi.extras/host-l
ibs-i686-mingw32/usr --with-gmp=/scratch/maciej/arm-eabi-2014.05-rel/obj/pkg-201
4.05-28-arm-none-eabi/arm-2014.05-28-arm-none-eabi.extras/host-libs-i686-mingw32
/usr --with-mpfr=/scratch/maciej/arm-eabi-2014.05-rel/obj/pkg-2014.05-28-arm-non
e-eabi/arm-2014.05-28-arm-none-eabi.extras/host-libs-i686-mingw32/usr --with-mpc
=/scratch/maciej/arm-eabi-2014.05-rel/obj/pkg-2014.05-28-arm-none-eabi/arm-2014.
05-28-arm-none-eabi.extras/host-libs-i686-mingw32/usr --with-isl=/scratch/maciej
/arm-eabi-2014.05-rel/obj/pkg-2014.05-28-arm-none-eabi/arm-2014.05-28-arm-none-e
abi.extras/host-libs-i686-mingw32/usr --with-cloog=/scratch/maciej/arm-eabi-2014
.05-rel/obj/pkg-2014.05-28-arm-none-eabi/arm-2014.05-28-arm-none-eabi.extras/hos
t-libs-i686-mingw32/usr --disable-libgomp --disable-libitm --disable-libatomic -
-disable-libssp --enable-poison-system-directories --with-build-time-tools=/scra
tch/maciej/arm-eabi-2014.05-rel/obj/tools-i686-pc-linux-gnu-2014.05-28-arm-none-
eabi-i686-mingw32/arm-none-eabi/bin --with-build-time-tools=/scratch/maciej/arm-
eabi-2014.05-rel/obj/tools-i686-pc-linux-gnu-2014.05-28-arm-none-eabi-i686-mingw
32/arm-none-eabi/bin SED=sed
Thread model: single
gcc version 4.8.3 20140320 (prerelease) (Sourcery CodeBench Lite 2014.05-28)