RT-Thread/rt-thread #1685: 编写看门狗驱动时，看门狗驱动框架非要自动关闭看门狗吗？

liuI, 2018-08-01:
##描述
看门狗驱动编写完成，实现了看门狗驱动框架的所有的命令功能，包括开启、关闭看门狗。
但当我把这个系统部署后，却发现看门狗没有起作用，死机了不进行重启，反复检查驱动发现没有问题；
后来定位到问题为：**在应用层调用一次rt_device_close函数关闭设备句柄后，函数内部会给驱动发送一个RT_DEVICE_CTRL_WDT_STOP指令一并关闭看门狗！**
##问题来了
驱动框架这么设计的初衷是什么？为什么要在关闭句柄时、把看门狗也关掉？
我查看了一下stm32f10x的看门狗驱动的例子，这个并没有实现RT_DEVICE_CTRL_WDT_STOP命令，我可以理解为这个例子是碰巧躲过了一劫？还是说，本来应用层就必须要保证这个设备驱动的句柄一直保存打开状态？但这样做，在设计上难道不是存在着问题？
###我暂时的临时解决办法
注释掉了RT_DEVICE_CTRL_WDT_STOP功能

BernardXiong, 2018-08-08:
有可能这里有些不一样的，后面给一份WDT管理框架吧

XXXXzzzz000, 2018-10-25:
@BernardXiong 我之前看到esp32里面的看门狗的话是直接在menuconfig中配置好相关参数就可以用，是不是可以考虑下这种方式。

mysterywolf, 2021-02-15:
STM32的看门狗开了就关不了

mysterywolf, 2021-02-15:
你是调用rt_device_close把看门狗给关了嘛？

Guozhanxin, 2021-02-24:
> 你是调用rt_device_close把看门狗给关了嘛？

[image]
看起来是的

closed 2021-02-24 by mysterywolf