RT-Thread/rt-thread #1691: AT 组件 at_client_getchar()有一定概率返回无效字符

MarcusXing, 2018-08-03:
现象：
[E/AT] Read response buffer failed. The Response buffer size is out of buffer size(64)!

可能的原因：
rt_sem_take之后，存在rt_device_read返回值为0的情况，引起return ch;返回了无效的字符。
无效字符过多超过buffer size。

armink, 2018-08-03:
好的，晚些时候修一下，多谢反馈哈

armink, 2018-08-04:
已经在 #1693 中修复

closed 2018-08-04 by armink

armink, 2018-08-04:
所以，你测试的还是存在这种情况

armink, 2018-08-04:
你改成这样试试吧
```c
static char at_client_getchar(void)
{
    char ch;

    while (rt_device_read(at_client_local->device, 0, &ch, 1) == 0)
    {
        rt_sem_take(at_client_local->rx_notice, RT_WAITING_FOREVER);
    }

    return ch;
}
```

MarcusXing, 2018-08-04:
@armink
这个可以，不过我有点担心效率问题
增加了次数检测之后
static char at_client_getchar(void)
{
    char ch;
    unsigned long times = 0;

    while (rt_device_read(at_client_local->device, 0, &ch, 1) == 0)
    {
        rt_sem_take(at_client_local->rx_notice, RT_WAITING_FOREVER);
        times++;
    }
    if(times) LOG_D("rt_device_read retry %d times", times);
    return ch;
}
有时候while循环比较多：
[D/AT] rt_device_read retry 38 times

armink, 2018-08-04:
OK，你再试试这个方式？

```
static char at_client_getchar(void)
{
    char ch;

    while (rt_device_read(at_client_local->device, 0, &ch, 1) == 0)
    {

        rt_sem_take(at_client_local->rx_notice, RT_WAITING_FOREVER);
        rt_sem_control(at_client_local->rx_notice, RT_IPC_CMD_RESET, RT_NULL);
    }

    return ch;
}
```

MarcusXing, 2018-08-04:
目前看重试次数都是在2次以内了

\ | /
- RT -     Thread Operating System
 / | \     3.1.0 build Aug  4 2018
 2006 - 2018 Copyright by rt-thread team
[I/AT] RT-Thread AT server (V0.2.3) initialize success.
[I/AT] RT-Thread AT client (V0.2.3) initialize success.
[I/SAL_SOC] Socket Abstraction Layer initialize success.
[D/AT] Start initializing the EC20 module
[D/AT] send: 0000-0020: 41 54                                                                                                 AT
[D/AT] rt_device_read retry 1 times
[D/AT] recvline: 0000-0020: 41 54 0D 0D 0A                                                                                        AT...
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] send: 0000-0020: 41 54 56 31                                                                                           ATV1
[D/AT] rt_device_read retry 2 times
[D/AT] recvline: 0000-0020: 41 54 56 31 0D 0D 0A                                                                                  ATV1...
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] send: 0000-0020: 41 54 45 30                                                                                           ATE0
[D/AT] rt_device_read retry 2 times
[D/AT] recvline: 0000-0020: 41 54 45 30 0D 0D 0A                                                                                  ATE0...
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] send: 0000-0020: 41 54 2B 43 4D 45 45 3D  32                                                                           AT+CMEE=2
[D/AT] rt_device_read retry 2 times
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] send: 0000-0020: 41 54 2B 49 50 52 3F                                                                                  AT+IPR?
[D/AT] rt_device_read retry 2 times
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 2B 49 50 52 3A 20 31 31  35 32 30 30 0D 0A                                                            +IPR: 115200..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] Baudrate 115200
[D/AT] send: 0000-0020: 41 54 49                                                                                              ATI
[D/AT] rt_device_read retry 2 times
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 51 75 65 63 74 65 6C 0D  0A                                                                           Quectel..
[D/AT] recvline: 0000-0020: 45 43 32 30 46 0D 0A                                                                                  EC20F..
[D/AT] recvline: 0000-0020: 52 65 76 69 73 69 6F 6E  3A 20 45 43 32 30 43 45  46 44 4B 47 52 30 36 41  30 34 4D 32 47 0D 0A       Revision: EC20CEFDKGR06A04M2G..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..

armink, 2018-08-04:
好的，调整了下顺序，应该有效果，麻烦再试试哈
```c
static char at_client_getchar(void)
{
    char ch;

    while (rt_device_read(at_client_local->device, 0, &ch, 1) == 0)
    {
        rt_sem_control(at_client_local->rx_notice, RT_IPC_CMD_RESET, RT_NULL);
        rt_sem_take(at_client_local->rx_notice, RT_WAITING_FOREVER);
    }

    return ch;
}
```

MarcusXing, 2018-08-04:
这次好了

armink, 2018-08-04:
好的，代码已经更新了，多谢反馈哈~