RT-Thread/rt-thread #1712: AT组件Enable print RAW format AT command communication data会丢数据

MarcusXing, 2018-08-11:
现象：
[D/AT] send: 0000-0020: 41 54 2B 51 49 44 4E 53  47 49 50 3D 31 2C 22 62  61 69 64 75 2E 63 6F 6D  22                         AT+QIDNSGIP=1,"baidu.com"
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 4F 4B 0D 0A                                                                                           OK..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 6E 73 67 69 70 22 2C 22  32 32 30 2E 31 38 31 2E  35 37 2E 32 31 36 22 0D  0A                         nsgip","220.181.57.216"..
[D/AT] recvline: 0000-0020: 0D 0A                                                                                                 ..
[D/AT] recvline: 0000-0020: 2B 51 49 55 52 43 3A 20  22 64 6E 73 67 69 70 22  2C 22 31 32 33 2E 31 32  35 2E 31 31 35 2E 31 31    +QIURC: "dnsgip","123.125.115.11
[D/AT] recvline: 0020-0040: 30 22 0D 0A      
电脑串口抓的实际数据流：
RXD：
AT+QIDNSGIP=1,"baidu.com"
TXD：
OK

+QIURC: "dnsgip",0,2,399

+QIURC: "dnsgip","220.181.57.216"

+QIURC: "dnsgip","123.125.115.110"

丢失的部分：
+QIURC: "dnsgip",0,2,399

+QIURC: "d

经验证在：
static int at_recv_readline(void)
的
        ch = at_client_getchar();
后面增加
        rt_kprintf("%c", ch);
也不会打印出丢失的部分

然而如果关掉Enable print RAW format AT command communication data
则不会丢数据

原因推测：
由于DNS被本地缓存，因此该指令返回特别快，而且数据量较大，可能存在无法及时接收的情况。

armink, 2018-08-11:
finsh 串口的波特率有没有提高，比如到 921600 ，不提高的话，大数据量输出时，会阻塞 AT 串口

armink, 2018-08-14:
还有问题吗

MarcusXing, 2018-08-14:
关掉吧，虽然我觉得这块还可以再优化一下，毕竟mcu还要干别的活，finsh不影响，别的用户应用也还是有可能影响。

armink, 2018-08-14:
毕竟串口带宽有限，两个思路

- 修改波特率
- 修改串口缓冲区大小

MarcusXing, 2018-08-14:
等我有时间，试试DMA收发

closed 2018-08-14 by armink