RT-Thread/rt-thread #1734: linux下编译qemu-vexpress-a9出错

haowenchao, 2018-08-23:
Ubuntu18.04环境rt-thread/bsp/qemu-vexpress-a9目录下执行scons报错：

/train/rt-thread/components/dfs/include/dfs_select.h:10:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                      ^
......
scons: building terminated because of errors.

根据错误信息，未找到fd_set类型

试验：
1、直接使用命令预编译：
cd /train/rt-thread/components/dfs/include
arm-none-eabi-gcc -I /train/rt-thread/include -I /train/rt-thread/bsp/qemu-vexpress-a9/ -E dfs_select.h
根据输出成功把fd-set替换为_types_fd_set
2、用1）中预编译得到的结果替换dfs_select.h，这部分代码编译通过，但在后面的编译中多处报错，一个是_types_fd_set在多出定义，另一个是fd_set未定义。

附件是1）中预处理之后的到文件。

BernardXiong, 2018-08-23:
把bsp/qemu-vexpress-a9下的cconfig.h文件删除，或者执行一遍`scons -c`，然后再执行`scons`看看是否正确。

如果错误依旧，把你的工具链情况说下。

haowenchao, 2018-08-23:
https://github.com/RT-Thread/rt-thread/issues/1734#issuecomment-415387073
按照您说的方法，问题解决。

closed 2018-08-28 by BernardXiong

tecty, 2019-08-27:
我也遇到了同样的问题。没法通过`scons-c` 或者删除`ccconf` 解决

```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/6.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:6.3.1+svn253039-1build1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1) 
```

tecty, 2019-08-27:
我在论坛中找到，手动定义一个宏的确可以过编译
> 那怎么在RTT3中使用menuconfig？env中的menuconfig指的是什么？BTW。 我手动在rtconfig.h中加人 #define HAVE_SYS_SELECT_H  就能编译过了