RT-Thread/rt-thread #1769: stm32f1_wdg.c驱动建议

rockjg0917, 2018-09-03:
rt_uint32_t timeout = 1000;
rt_device_control(watchdog_device, RT_DEVICE_CTRL_WDT_SET_TIMEOUT, (rt_uint32_t*)timeout);
static rt_err_t stm32f1_wdg_control(rt_watchdog_t *wdt, int cmd, void *arg)
{
     rt_uint32_t timeout_ms = 0;
     rt_uint32_t timeout_pow = 1;
     switch(cmd)
     {
     case RT_DEVICE_CTRL_WDT_SET_TIMEOUT:
          timeout_ms = (rt_uint32_t) arg;
**上面是原来的驱动，timeout参数值需要转换为指针，但不是其地址，改成如下会不会好点？主要是传参习惯**
rt_uint32_t timeout = 1000;
rt_device_control(watchdog_device, RT_DEVICE_CTRL_WDT_SET_TIMEOUT, &timeout);
static rt_err_t stm32f1_wdg_control(rt_watchdog_t *wdt, int cmd, void *arg)
{
     rt_uint32_t timeout_ms = 0;
     rt_uint32_t timeout_pow = 1;
     switch(cmd)
     {
     case RT_DEVICE_CTRL_WDT_SET_TIMEOUT:
          //timeout_ms = (rt_uint32_t) arg;
	timeout_ms = *(rt_uint32_t*) arg;

uestczyh222, 2018-09-03:
你这不就bug了吗，void*是一个无明确意义的四字节变量，直接传会不会好一些

uestczyh222, 2018-09-03:
你可以直接rt_device_control(watchdog_device, RT_DEVICE_CTRL_WDT_SET_TIMEOUT,1000);
生成的汇编代码都要少上几行

rockjg0917, 2018-09-03:
写错了，应该是timeout_ms = * (rt_uint32_t *) arg;

rockjg0917, 2018-09-03:
rt_err_t rt_device_control(rt_device_t dev, int cmd, void *arg)，接口是固定的啊，传入的就是指针类型

uestczyh222, 2018-09-03:
传入的是32位变量

rockjg0917, 2018-09-03:
嗯，一开始没看里面源码，看形参是指针，下意识传入变量地址，后来发现有问题，需要传入32位变量，不算bug，我只是提一下建议。

JiangHeng12138, 2018-09-04:
这谁看了都会觉得应该传指针

closed 2021-02-05 by mysterywolf