RT-Thread/rt-thread #1806: 使用自带的RTT+aliot+ESP8266无法连接MQTT，SSL/TLS会出现

misdangerous, 2018-09-14:
msh />ali_mqtt_test start
[dbg] iotx_device_info_init(32): device_info already created, return!
[dbg] iotx_device_info_set(50): start to set device info!
[dbg] iotx_device_info_set(64): device_info set successfully!
[dbg] guider_print_dev_guider_info(271): ....................................................
[dbg] guider_print_dev_guider_info(272):           ProductKey : a1K6ElXzgfQ
[dbg] guider_print_dev_guider_info(273):           DeviceName : LP-TEST-DEV-1
[dbg] guider_print_dev_guider_info(274):             DeviceID : a1K6ElXzgfQ.LP-TEST-DEV-1
[dbg] guider_print_dev_guider_info(276): ....................................................
[dbg] guider_print_dev_guider_info(277):        PartnerID Buf : ,partner_id=example.demo.partner-id
[dbg] guider_print_dev_guider_info(278):         ModuleID Buf : ,module_id=example.demo.module-id
[dbg] guider_print_dev_guider_info(279):           Guider URL : 
[dbg] guider_print_dev_guider_info(281):       Guider SecMode : 2 (TLS + Direct)
[dbg] guider_print_dev_guider_info(283):     Guider Timestamp : 2524608000000
[dbg] guider_print_dev_guider_info(284): ....................................................
[dbg] guider_print_dev_guider_info(290): ....................................................
[dbg] guider_print_conn_info(248): -----------------------------------------
[dbg] guider_print_conn_info(249):             Host : a1K6ElXzgfQ.iot-as-mqtt.cn-shanghai.aliyuncs.com
[dbg] guider_print_conn_info(250):             Port : 1883
[dbg] guider_print_conn_info(253):         ClientID : a1K6ElXzgfQ.LP-TEST-DEV-1|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=example.demo.partner-id,module_id=example.demo.module-id|
[dbg] guider_print_conn_info(255):       TLS PubKey : 08071c6c ('-----BEGIN CERTI ...')
[dbg] guider_print_conn_info(258): -----------------------------------------
host: a1k6elxzgfq.iot-as-mqtt.cn-shanghai.aliyuncs.com
[inf] iotx_mc_init(1703): MQTT init success!
[inf] _ssl_client_init(175): Loading the CA root certificate ...
cert. version     : 3
serial number     : 04:00:00:00:00:01:15:4B:5A:C3:94
issuer name       : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
subject name      : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
issued  on        : 1998-09-01 12:00:00
expires on        : 2028-01-28 12:00:00
signed using      : RSA with SHA1
RSA key size      : 2048 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sign
[inf] _ssl_parse_crt(143): crt content:451
[inf] _ssl_client_init(183):  ok (0 skipped)
[inf] _TLSConnectNetwork(345): Connecting to /a1k6elxzgfq.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
msh />[inf] _TLSConnectNetwork(358):  ok
[inf] _TLSConnectNetwork(363):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(373):  ok
[inf] _TLSConnectNetwork(408): Performing the SSL/TLS handshake...
[inf] _TLSConnectNetwork(412): failed  ! mbedtls_ssl_handshake returned -0x4290
[inf] _network_ssl_disconnect(514): ssl_disconnect
[inf] HAL_SSL_Destroy(530): handle is NULL
[err] iotx_mc_connect(2001): TCP or TLS Connection failed
[inf] HAL_SSL_Destroy(530): handle is NULL
[inf] iotx_mc_disconnect(2129): mqtt disconnect!
[inf] iotx_mc_release(2175): mqtt release!
mqtt_client|273 :: MQTT construct failed
[err] LITE_dump_malloc_free_stats(594): WITH_MEM_STATS = 0
mqtt_client|344 :: out of sample!

打开SSL/TLS的调试信息

packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2416: => flush output
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2434: message length: 7, out_left: 7
[E/AT_SOC] send data error, current socket (0) state (3) is error.
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2441: ssl->f_send() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2846: mbedtls_ssl_flush_output() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:4134: mbedtls_ssl_write_record() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_cli.c:2598: mbedtls_pk_verify() returned -17040 (-0x4290)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:6567: <= handshake
[inf] _TLSConnectNetwork(412): failed  ! mbedtls_ssl_handshake returned -0x4290
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:7186: => write close notify
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2416: => flush output
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2434: message length: 7, out_left: 7
[E/AT_SOC] send data error, current socket (0) state (3) is error.
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2441: ssl->f_send() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:7344: => free
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:7409: <= free
[inf] _network_ssl_disconnect(514): ssl_disconnect
发现好像是因为服务器断开了socket，想请教一下什么问题

HubertXie, 2018-09-15:
我遇到过，是at_socket的问题

Lawlieta, 2018-09-17:
问题已经修复，请更新 RT-Thread 和 at device 最新源码测试

misdangerous, 2018-09-17:
能请教一下是什么地方问题吗？

Lawlieta, 2018-09-17:
应该是 AT Socket 中判断 select send 和 recv 事件时候的问题，ali-iotkit 中的 MQTT 连接中的 send 和 recv 操作都要使用select，原来的 AT Socket 对 select 中的 send event 事件处理有些问题，导致发送数据失败无法连接服务器

misdangerous, 2018-09-20:
这两天我测试了一下，连接上阿里云物联网没有问题，但是打开SSL/TLS之后还是会报错
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:4125: send alert level=2 message=51
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2701: => write record
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2835: output record: msgtype = 21, version = [3:3], msglen = 2
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2840: dumping 'output record sent to network' (7 bytes)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2840: 0000:  15 03 03 00 02 02 33                             ......3
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2416: => flush output
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2434: message length: 7, out_left: 7
[E/AT_SOC] send data error, current socket (0) state (3) is error.
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2441: ssl->f_send() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2846: mbedtls_ssl_flush_output() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:4134: mbedtls_ssl_write_record() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_cli.c:2598: mbedtls_pk_verify() returned -17040 (-0x4290)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:6567: <= handshake
[inf] _TLSConnectNetwork(412): failed  ! mbedtls_ssl_handshake returned -0x4290
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:7186: => write close notify
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2416: => flush output
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2434: message length: 7, out_left: 7
[E/AT_SOC] send data error, current socket (0) state (3) is error.
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:2441: ssl->f_send() returned -78 (-0x004e)
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:7344: => free
packages\mbedtls-latest\mbedtls\library\ssl_tls.c:7409: <= free
[inf] _network_ssl_disconnect(514): ssl_disconnect
[inf] HAL_SSL_Destroy(530): handle is NULL
[err] iotx_mc_connect(2001): TCP or TLS Connection failed
[inf] HAL_SSL_Destroy(530): handle is NULL
[inf] iotx_mc_disconnect(2129): mqtt disconnect!
[inf] iotx_mc_release(2175): mqtt release!
mqtt_client|273 :: MQTT construct failed
[err] LITE_dump_malloc_free_stats(594): WITH_MEM_STATS = 0
mqtt_client|344 :: out of sample!

closed 2022-08-06 by mysterywolf