RT-Thread/rt-thread #1841: "rt_uint32_t与uint32_t定义不同"问题的建议

liruncong, 2018-09-24:
问题: uint_32_t为C99标准定义(C99标准新增标准头文件stdint.h),与rt_uint32_t定义不同. 导致使用uint_32类型传给使用rt_uint32_t的内核函数时armclang编译器给出错误.
建议: 将现有基本类型定义
_________________________________________________________________
/* RT-Thread basic data type definitions */
typedef signed   char                   rt_int8_t;      /**<  8bit integer type */
typedef signed   short                  rt_int16_t;     /**< 16bit integer type */
typedef signed   long                   rt_int32_t;     /**< 32bit integer type */
typedef unsigned char                   rt_uint8_t;     /**<  8bit unsigned integer type */
typedef unsigned short                  rt_uint16_t;    /**< 16bit unsigned integer type */
typedef unsigned long                   rt_uint32_t;    /**< 32bit unsigned integer type */
typedef int                             rt_bool_t;      /**< boolean type */
__________________________________________________________________
修改为:
__________________________________________________________________
typedef int8_t                          rt_int8_t;      /**<  8bit integer type */
typedef int16_t                         rt_int16_t;     /**< 16bit integer type */
typedef int32_t                         rt_int32_t;     /**< 32bit integer type */
typedef uint8_t                         rt_uint8_t;     /**<  8bit unsigned integer type */
typedef uint16_t                        rt_uint16_t;    /**< 16bit unsigned integer type */
typedef uint32_t                        rt_uint32_t;    /**< 32bit unsigned integer type */
typedef int32_t                         rt_bool_t;      /**< boolean type */
__________________________________________________________________
这个可以直接编译通过,无需其它更改.
注: 编译器会自动根据当前机器决定uint32_t使用unsigned int还是unsigned long. 通常时unsigned int如果是32位的,那么就定义uint32_t为unsigned int,否则定义为unsigned long.

liruncong, 2018-09-25:
@BernardXiong 我测试了以下情形.
1) clang(从http://releases.llvm.org/download.html#7.0.0下载). 预处理文件给出以下定义(节选)
____________________________________________________________________________________________________
typedef signed char int8_t;
typedef short int16_t;
typedef int int32_t;
typedef long long int64_t;
typedef unsigned char uint8_t;
typedef unsigned short uint16_t;
typedef unsigned int uint32_t;                                     // ***
typedef unsigned long long uint64_t;
____________________________________________________________________________________________________
注: 这个clang就是目前armclang用的前端编译器
2) vs2017中编写以下代码
____________________________________________________________________________________________________
void test(uint32_t* pTest) {

}
int main()
{
    unsigned long ul;                     // 出现错误, 修改为unsigned int ul;后正常
    test(&ul);
}
______________________________________________________________________________________________________
出现错误: E0167	"unsigned long *" 类型的实参与 "uint32_t *" 类型的形参不兼容

另外, armclang需要屏蔽一些警告,我的armclang PR中的rtconfig.py设定如下:
_____________________________________________________________
    CFLAGS += ' -fshort-wchar -fdiagnostics-format=msvc -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables'
    CFLAGS += ' -Wno-nonportable-include-path -Wno-invalid-source-encoding -Wunused-variable -Wformat -Wuninitialized -Wfloat-conversion'
    CFLAGS += ' -Wno-unicode-homoglyph -Wno-unused-command-line-argument -Wno-reserved-user-defined-literal -Wno-unused-const-variable'
    CFLAGS += ' -Wno-incompatible-library-redeclaration'
_______________________________________________________________
这样基本上没有警告了.

BernardXiong, 2018-10-06:
现在看下来，uint32确实应该是使用unsigned int。这个在64位移植中也发现了，后面再改正过来。

closed 2018-10-06 by BernardXiong