RT-Thread/rt-thread #1843: 关于提高rt_memcpy等函数效率的建议

liruncong, 2018-09-25:
目前rt_memcpy函数:
_____________________________________________________________________
void *rt_memcpy(void *dst, const void *src, rt_ubase_t count)
{
.....
}
.....
#if !defined (RT_USING_NEWLIB) && defined (RT_USING_MINILIBC) && defined (__GNUC__)
#include <sys/types.h>
void *memcpy(void *dest, const void *src, size_t n) __attribute__((weak, alias("rt_memcpy")));
.....
_____________________________________________________________________
更改为:
#include <string.h>
#if !defined (RT_USING_NEWLIB) && defined (RT_USING_MINILIBC) && defined (__GNUC__)
void *memcpy(void *dst, const void *src, rt_ubase_t count)
#else
RT_WEAK void *memcpy(void *dst, const void *src, rt_ubase_t count)
#endif 
{
.....//此处内容为原rt_memcpy内容
}
_____________________________________________________________________
注: 在ROM容量极小时,使用自定义memcpy覆盖掉gcc libc中的memcpy. 其它情况使用编译器c库函数

BernardXiong, 2018-09-26:
直接实现为memcpy？这里面的需求各式各样，不能一概而论啊

liruncong, 2018-09-26:
不就两种情况吗？一种是使用自定义的拷贝函数，一种是调用c库函数。还有别的吗?

BernardXiong, 2018-09-26:
还有独立一个单一内核，保持这个核心尽量小；然后外围的组件，应用又能够用libc的高速memcpy接口

系统启动时，先加载独立核心（从SPI Flash加载到DRAM，要保持核心尽量小，这样才能够做到启动速度尽量快），然后再加载外围组件，应用。

liruncong, 2018-09-26:
是不是要两次编译成两个二进制文件，第1个用于初始化内存等，并把第2个加载到内存执行。第2个才是主程序。是这样吗？

BernardXiong, 2018-09-26:
就一个，分散加载。可以了解下

liruncong, 2018-09-26:
// .h文件
#if 特殊应用
typedef void (*rt_memcpy_fun_ptr)(void *dst, const void *src, rt_ubase_t count);
extern rt_memcpy_fun_ptr rt_memcpy;
#else
#define rt_memcpy memcpy
#endif

// .c文件
#if 特殊应用
// 1. 微内核加载区
void _rt_memcpy(void *dst, const void *src, rt_ubase_t count) // 微内核用版本
{
    ....
}
// 默认rt_memcpy(..)调用_rt_memcpy. 
// 在微内核结束前设置rt_memcpy=memcpy. memcpy加载在非微内核区
rt_memcpy_fun_ptr rt_memcpy = _rt_memcpy; 
#endif
@BernardXiong 你看看这样是否可行?

heyuanjie87, 2018-09-28:
我用rtmemcpy是为了避开mdk的非对齐时的bug

BernardXiong, 2018-09-29:
所以完全替换掉，不用，几乎是不可能的任务。

BernardXiong, 2018-09-29:
有的时候一直存在，而想了好久都未拿掉，会有它存在的合理的一面。

closed 2018-09-29 by BernardXiong

liruncong, 2018-09-29:
个人觉得如果有heyuanjie87这种情况，应该在他的bsp中重写memcpy等函数覆盖库函数。分散加载其实也应该在bsp中处理。libcpu启动代码应该留下weak函数接口给bsp使用，用于快速启动情形。对有cache和buffer的cpu，在搬运数据前应该开启cache和buffer（mmu放后面再启动），这个在am335x上测试过可以让zi段清零效率提高40%左右

liruncong, 2018-09-29:
个人觉得如果有heyuanjie87这种情况，应该在他的bsp中重写memcpy等函数覆盖库函数。分散加载其实也应该在bsp中处理。libcpu启动代码应该留下weak函数接口给bsp使用，用于快速启动情形。对有cache和buffer的cpu，在搬运数据前应该开启cache和buffer（mmu放后面再启动），这个在am335x上测试过可以让zi段清零效率提高40%左右。@BernardXiong