RT-Thread/rt-thread #1857: lwip1.4 中eth_netif_device_init初始化有问题

s05427226, 2018-09-30:
if(!(ethif->flags & ETHIF_LINK_PHYUP))
        {
            /* set link_up for this netif */
            netif_set_link_up(ethif->netif);
        }
逻辑上感觉应该改成
 if(ethif->flags & ETHIF_LINK_PHYUP)
        {
            /* set link_up for this netif */
            netif_set_link_up(ethif->netif);
        }

BernardXiong, 2018-10-06:
ethif->flags已经link up了，下面的netif_set_link_up就不需要执行了。正式因为它没有link up，所以下面才强制的link up起来。

s05427226, 2018-10-08:
这个地方的代码是否可以注释掉，因为在phy_monitor_thread_entry 这个线程中，会去检测linkdown/up的状态

BernardXiong, 2018-10-08:
两个设备怎么会用一个地址呢？

可以系统启动后，自己主动把一个网络接口处理下。

s05427226, 2018-10-08:
netif_set_link_up 函数中会设置netif->flags |= NETIF_FLAG_LINK_UP;phy_monitor_thread_entry  检测到网线插拔的时候，也会调用netif_set_link_up，但是netif->flags & NETIF_FLAG_LINK_UP这个时候是1，就不会调用if里面的代码，所以导致没有发送gratuitous arp

s05427226, 2018-10-08:
另外，如果两个设备是同一个IP，会在arp表中有缓存，这个时候就会有个超时或者老化时间，导致第二个设备要过很长时间才能通信上

mysterywolf, 2022-03-05:
@xiangxistu 贤良，检查一下新版整合之后，此处还需要改嘛？

xiangxistu, 2022-03-25:
这个应该不会，任何尝试发送新的报文，在 ARP 表中为空时;

lwip 都会阻塞任何发送，先去发送 ARP 请求的，获取到 ARP reply 后才能有发送动作的。
上面说的 LINK_UP 的逻辑问题，我们这边不需要改了。现在对这个状态的侦测，挺合理的。

closed 2022-03-28 by mysterywolf