RT-Thread/rt-thread #1860: SAL组件MQTT里面的select函数超时问题

MarcusXing, 2018-10-03:
现象
SAL组件MQTT里面的select函数有时候不按设定的超时时间工作，在没有数据的情况下，可能提前返回
可能的原因
尚不清楚

BernardXiong, 2018-10-06:
@Lawlieta Please take a look.

Lawlieta, 2018-10-08:
@MarcusXing 你好，关于select 超时问题，我这边使用 MQTT 软件包示例和官方 TCP select 示例测试，发现并没有出现超时时间和设置时间不一样的问题 ，下图是测试结果：
[image]
[image]
请你那边给出更多测试环境和数据确定问题，谢谢。

MarcusXing, 2018-11-15:
我当时是拿MQTT组件的代码进行大范围的修改，形成自定义协议的时候发现的这个问题。
我没有通过select来访问网络接口，而是只用于访问pipe。
我在while循环中使用select不断的查询pipe里是否有内容。
一开始超时设置是正确的，当我从pipe中成功读取到一次数据之后，接下来的select调用都会立即返回。然后读取pipe的函数只读到0字节。由于pipe中的数据量未知，我每次都是循环读到pipe空为止。

MarcusXing, 2018-11-15:
sorry过了这么多天才反馈。
我注册github用的微软邮箱比较垃圾，直到今天我才注意到有新邮件提醒。
我是新接触pipe这个组件，在读写pipe的时候遗漏了什么也是有可能的。
这个问题在我的代码里是必然出现的，并不是概率问题。
我的临时解决方法是开了一个事件，在写pipe之后额外发送一个事件。在MQTT循环中等待事件，再去select

closed 2022-07-28 by mysterywolf