RT-Thread/rt-thread #1888: GNU GCC & newlib version issue

BernardXiong, 2018-10-12:
In Linux environment, RT-Thread will have restrictions on version of GCC and newlib. For example, only a few versions of GNU Arm Embedded Toolchain are verified now:
GNU Arm Embedded Toolchain 5.4, 6. And not support GNU Arm Embedded Toolchain 7.

But for the new version, or the toolchains self builds, it does not support well.

BernardXiong, 2018-11-10:
fix with #1964 except for pthreads.

pinxue, 2019-11-07:
The newlib-pthread error still exists.

toolchain:
arm-none-eabi-binutils         @2.33.1         cross/arm-none-eabi-binutils
arm-none-eabi-gcc              @9.2.0          cross/arm-none-eabi-gcc
arm-none-eabi-gdb              @7.9            cross/arm-none-eabi-gdb

error message:
10:qemu-vexpress-a9 pinxue$ scons
scons: Reading SConscript files ...
b''
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/lcd_init.o
CC build/applications/main.o
CC build/applications/mnt.o
CC build/drivers/board.o
CC build/drivers/drv_keyboard.o
CC build/drivers/drv_mouse.o
CC build/drivers/drv_sdio.o
CC build/drivers/drv_smc911x.o
CC build/drivers/drv_timer.o
CC build/drivers/secondary_cpu.o
CC build/drivers/serial.o
CXX build/kernel/components/cplusplus/Mutex.o
CXX build/kernel/components/cplusplus/Semaphore.o
CXX build/kernel/components/cplusplus/Thread.o
CXX build/kernel/components/cplusplus/crt.o
CC build/kernel/components/cplusplus/crt_init.o
CC build/kernel/components/dfs/filesystems/devfs/devfs.o
CC build/kernel/components/dfs/filesystems/elmfat/dfs_elm.o
CC build/kernel/components/dfs/filesystems/elmfat/ff.o
CC build/kernel/components/dfs/filesystems/elmfat/option/ccsbcs.o
CC build/kernel/components/dfs/filesystems/ramfs/dfs_ramfs.o
CC build/kernel/components/dfs/filesystems/romfs/dfs_romfs.o
CC build/kernel/components/dfs/filesystems/romfs/romfs.o
CC build/kernel/components/dfs/src/dfs.o
CC build/kernel/components/dfs/src/dfs_file.o
CC build/kernel/components/dfs/src/dfs_fs.o
CC build/kernel/components/dfs/src/dfs_posix.o
CC build/kernel/components/dfs/src/poll.o
CC build/kernel/components/dfs/src/select.o
CC build/kernel/components/drivers/i2c/i2c-bit-ops.o
CC build/kernel/components/drivers/i2c/i2c_core.o
CC build/kernel/components/drivers/i2c/i2c_dev.o
CC build/kernel/components/drivers/misc/pin.o
CC build/kernel/components/drivers/mtd/mtd_nand.o
CC build/kernel/components/drivers/mtd/mtd_nor.o
CC build/kernel/components/drivers/rtc/rtc.o
CC build/kernel/components/drivers/rtc/soft_rtc.o
CC build/kernel/components/drivers/sdio/block_dev.o
CC build/kernel/components/drivers/sdio/mmc.o
CC build/kernel/components/drivers/sdio/mmcsd_core.o
CC build/kernel/components/drivers/sdio/sd.o
CC build/kernel/components/drivers/sdio/sdio.o
CC build/kernel/components/drivers/serial/serial.o
CC build/kernel/components/drivers/spi/sfud/src/sfud.o
~rt-thread/components/drivers/spi/sfud/src/sfud.c: In function 'set_write_enabled':
~rt-thread/components/drivers/spi/sfud/src/sfud.c:900:77: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
  900 |         } else if (!enabled && (register_status & SFUD_STATUS_REGISTER_WEL) == 1) {
      |                                                                             ^~
CC build/kernel/components/drivers/spi/sfud/src/sfud_sfdp.o
CC build/kernel/components/drivers/spi/spi_core.o
CC build/kernel/components/drivers/spi/spi_dev.o
CC build/kernel/components/drivers/spi/spi_flash_sfud.o
CC build/kernel/components/drivers/spi/spi_msd.o
CC build/kernel/components/drivers/src/completion.o
CC build/kernel/components/drivers/src/dataqueue.o
CC build/kernel/components/drivers/src/pipe.o
[... 176 lines cut ...]
~rt-thread/components/libc/pthreads/pthread.h:109:29: error: conflicting types for 'pthread_cond_t'
  109 | typedef struct pthread_cond pthread_cond_t;
      |                             ^~~~~~~~~~~~~~
In file included from /opt/local/arm-none-eabi/include/sys/types.h:239,
                 from /opt/local/arm-none-eabi/include/sys/signal.h:12,
                 from ~rt-thread/include/libc/libc_signal.h:67,
                 from ~rt-thread/include/rtdef.h:466,
                 from ~rt-thread/include/rtthread.h:24,
                 from ~rt-thread/components/libc/compilers/newlib/libc.c:10:
/opt/local/arm-none-eabi/include/sys/_pthreadtypes.h:176:20: note: previous declaration of 'pthread_cond_t' was here
  176 | typedef __uint32_t pthread_cond_t;       /* identify a condition variable */
      |                    ^~~~~~~~~~~~~~
In file included from ~rt-thread/components/libc/compilers/newlib/libc.c:19:
~rt-thread/components/libc/pthreads/pthread.h:191:5: error: conflicting types for 'pthread_kill'
  191 | int pthread_kill(pthread_t thread, int sig);
      |     ^~~~~~~~~~~~
In file included from ~rt-thread/include/libc/libc_signal.h:67,
                 from ~rt-thread/include/rtdef.h:466,
                 from ~rt-thread/include/rtthread.h:24,
                 from ~rt-thread/components/libc/compilers/newlib/libc.c:10:
/opt/local/arm-none-eabi/include/sys/signal.h:226:5: note: previous declaration of 'pthread_kill' was here
  226 | int pthread_kill (pthread_t, int);
      |     ^~~~~~~~~~~~
scons: *** [build/kernel/components/libc/compilers/newlib/libc.o] Error 1
scons: building terminated because of errors.

Guozhanxin, 2021-09-28:
目前已支持 gcc10，这个issue可以关闭了

closed 2021-10-05 by BernardXiong

wahaha02, 2021-11-02:
@Guozhanxin 您好。在RTT v3.1.4上，GCC版本从5.4.1升级到10.2.1后，也遇到这个问题。请问如何修复？