RT-Thread/rt-thread #1914: RT-Thread input框架

BernardXiong, 2018-10-20:
@heyuanjie87 新提交了input框架 #1907 ，API主要包括如下：

```c
int rt_input_device_register(rt_input_t *dev, int devtype, const char *name);
int rt_input_device_unregister(rt_input_t *dev);

 int rt_input_handler_register(struct input_handler *handler);

int rt_input_device_open(struct input_handler *h);
int rt_input_device_close(struct input_handler *h);

int rt_input_report(rt_input_t *dev, int type, int code, int value);
int rt_input_rel_report(rt_input_t *dev, int code, int value);
int rt_input_abs_report(rt_input_t *dev, int code, int value);
int rt_input_key_report(rt_input_t *dev, int code, int value);
int rt_input_sync(rt_input_t *dev);

void _input_gettime(struct input_event *ev);
```

感觉在进行数据report时很奇异，定义非常模糊。以及input框架在整体的使用场合中，它在整体结构中处于什么样的位置也比较奇怪。这部分还需要分场景进行细化，讨论。

希望大家可以就input框架多进行讨论，最终有一份棒的input框架出来。

heyuanjie87, 2018-10-20:
先贴一下我的驱动和应用参考供大家讨论
#include <rtthread.h>
#include <rtdevice.h>

#include "board.h"

#include "fsl_gpio.h"
#include "fsl_lpi2c.h"

#ifdef INPUT_USING_TOUCH
#if 0
#define FTDEBUG      rt_kprintf
#else
#define FTDEBUG(...)
#endif

#define I2CBUS_NAME    "i2c1"
#define TOUCH_SLP_TIME          (RT_TICK_PER_SECOND * 5)

#define FT5206_TS_ADDR          0x38
#define TP_MAX_TOUCH_POINT      2

enum ft5x0x_ts_regs
{
    FT5X0X_REG_THGROUP                  = 0x80,
    FT5X0X_REG_THPEAK                   = 0x81,
    FT5X0X_REG_THCAL                    = 0x82,
    FT5X0X_REG_THWATER                  = 0x83,
    FT5X0X_REG_THTEMP                   = 0x84,
    FT5X0X_REG_THDIFF                   = 0x85,
    FT5X0X_REG_CTRL                     = 0x86,
    FT5X0X_REG_TIMEENTERMONITOR         = 0x87,
    FT5X0X_REG_PERIODACTIVE             = 0x88,
    FT5X0X_REG_PERIODMONITOR            = 0x89,
    FT5X0X_REG_HEIGHT_B                 = 0x8a,
    FT5X0X_REG_MAX_FRAME                = 0x8b,
    FT5X0X_REG_DIST_MOVE                = 0x8c,
    FT5X0X_REG_DIST_POINT               = 0x8d,
    FT5X0X_REG_FEG_FRAME                = 0x8e,
    FT5X0X_REG_SINGLE_CLICK_OFFSET      = 0x8f,
    FT5X0X_REG_DOUBLE_CLICK_TIME_MIN    = 0x90,
    FT5X0X_REG_SINGLE_CLICK_TIME        = 0x91,
    FT5X0X_REG_LEFT_RIGHT_OFFSET        = 0x92,
    FT5X0X_REG_UP_DOWN_OFFSET           = 0x93,
    FT5X0X_REG_DISTANCE_LEFT_RIGHT      = 0x94,
    FT5X0X_REG_DISTANCE_UP_DOWN         = 0x95,
    FT5X0X_REG_ZOOM_DIS_SQR             = 0x96,
    FT5X0X_REG_RADIAN_VALUE             = 0x97,
    FT5X0X_REG_MAX_X_HIGH               = 0x98,
    FT5X0X_REG_MAX_X_LOW                = 0x99,
    FT5X0X_REG_MAX_Y_HIGH               = 0x9a,
    FT5X0X_REG_MAX_Y_LOW                = 0x9b,
    FT5X0X_REG_K_X_HIGH                 = 0x9c,
    FT5X0X_REG_K_X_LOW                  = 0x9d,
    FT5X0X_REG_K_Y_HIGH                 = 0x9e,
    FT5X0X_REG_K_Y_LOW                  = 0x9f,
    FT5X0X_REG_AUTO_CLB_MODE            = 0xa0,
    FT5X0X_REG_LIB_VERSION_H            = 0xa1,
    FT5X0X_REG_LIB_VERSION_L            = 0xa2,
    FT5X0X_REG_CIPHER                   = 0xa3,
    FT5X0X_REG_G_MODE                   = 0xa4,
    FT5X0X_REG_PMODE                    = 0xa5, /* Power Consume Mode */
    FT5X0X_REG_FIRMID                   = 0xa6,
    FT5X0X_REG_STATE                    = 0xa7,
    FT5X0X_REG_VENDID                   = 0xa8,
    FT5X0X_REG_ERR                      = 0xa9,
    FT5X0X_REG_CLB                      = 0xaa,
};

#define CTRL_NOAUTO_MONITOR    0x00
[... 255 lines cut ...]
                last_y = ev.value;
        }
        else if (ev.type == EV_KEY)
        {
            if (ev.code == BTN_TOUCH)
                touch_down = ev.value;
        }
        else if (ev.type == EV_SYN)
        {
            rt_kprintf("%d %d %d\n", last_x, last_y, touch_down);
            retry = 0;
        }

        if (retry)
        {
            goto _again;
        }
    }

    data->point.x = last_x;
    data->point.y = last_y;
    data->state = (touch_down == RT_TRUE) ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;

    return false;
}

BernardXiong, 2018-10-20:
@heyuanjie87 代码好多，没有对齐没有高亮，看得好累。