RT-Thread/rt-thread #1993: [risc-v] hifive1 error on build

matassos, 2018-11-21:
Hello , 
I am trying to build using the newlib toolchain, without installing the FreedomStudio, since it uses the same "riscv64-unknown-elf-gcc". After setting EXEC_PATH   = '/opt/riscv/bin' and running scons, the following error appears. I found a related issue https://github.com/RT-Thread/rt-thread/issues/1156 but still cant find the problem and tried to translate the Chinese instructions to set the path, although I already edited it on the rtconfig.py file.
Thank you,
Tassos. 

> matassos@PC:~/Desktop/rt-thread-master/bsp/hifive1$ scons
> scons: Reading SConscript files ...
> scons: done reading SConscript files.
> scons: Building targets ...
> scons: building associated VariantDir targets: build
> CC build/applications/main.o
> In file included from /home/matassos/Desktop/rt-thread-master/include/libc/libc_signal.h:71,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtlibc.h:21,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtdef.h:1024,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtthread.h:23,
>                  from applications/main.c:27:
> /opt/riscv/riscv64-unknown-elf/include/sys/signal.h:42:7: error: redefinition of 'union sigval'
>  union sigval {
>        ^~~~~~
> In file included from /home/matassos/Desktop/rt-thread-master/include/rtlibc.h:21,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtdef.h:1024,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtthread.h:23,
>                  from applications/main.c:27:
> /home/matassos/Desktop/rt-thread-master/include/libc/libc_signal.h:31:7: note: originally defined here
>  union sigval
>        ^~~~~~
> In file included from /home/matassos/Desktop/rt-thread-master/include/libc/libc_signal.h:71,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtlibc.h:21,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtdef.h:1024,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtthread.h:23,
>                  from applications/main.c:27:
> /opt/riscv/riscv64-unknown-elf/include/sys/signal.h:47:8: error: redefinition of 'struct sigevent'
>  struct sigevent {
>         ^~~~~~~~
> In file included from /home/matassos/Desktop/rt-thread-master/include/rtlibc.h:21,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtdef.h:1024,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtthread.h:23,
>                  from applications/main.c:27:
> /home/matassos/Desktop/rt-thread-master/include/libc/libc_signal.h:39:8: note: originally defined here
>  struct sigevent
>         ^~~~~~~~
> In file included from /home/matassos/Desktop/rt-thread-master/include/libc/libc_signal.h:71,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtlibc.h:21,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtdef.h:1024,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtthread.h:23,
>                  from applications/main.c:27:
> /opt/riscv/riscv64-unknown-elf/include/sys/signal.h:72:3: error: conflicting types for 'siginfo_t'
>  } siginfo_t;
>    ^~~~~~~~~
> In file included from /home/matassos/Desktop/rt-thread-master/include/rtlibc.h:21,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtdef.h:1024,
>                  from /home/matassos/Desktop/rt-thread-master/include/rtthread.h:23,
>                  from applications/main.c:27:
> /home/matassos/Desktop/rt-thread-master/include/libc/libc_signal.h:58:24: note: previous declaration of 'siginfo_t' was here
>  typedef struct siginfo siginfo_t;
>                         ^~~~~~~~~
> scons: *** [build/applications/main.o] Error 1
> scons: building terminated because of errors.

BernardXiong, 2018-11-21:
Please use `scons -c` to clean object and `cconfig.h`, then use `scons` to build it.

matassos, 2018-11-21:
Thank you, 
this fixed the issue, although now appears another:

> ...
> CC build/kernel/src/signal.o
> CC build/kernel/src/thread.o
> CC build/kernel/src/timer.o
> LINK rtthread.elf
> /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fclose.o): ABI is incompatible with that of the selected emulation:
>   target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
> /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fclose.o)
> ...
> ...
> /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fclose.o): file class ELFCLASS64 incompatible with ELFCLASS32
> /opt/riscv/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: file in wrong format
> collect2: error: ld returned 1 exit status
> scons: *** [rtthread.elf] Error 1
> scons: building terminated because of errors.

Options are "riscv64-unknown-elf-" and "-march=rv32imac -mabi=ilp32". It is pretty standard to use riscv64 command and setting march and mabi to 32. I believe its the linker, I need to set emulation -m elf32lriscv but adding the option inside the LFLAGS (where the linker script is set) does not work. Setting it inside the linker script does not exist as an option, from my understanding after searching the documentation.

closed 2022-01-21 by mysterywolf