RT-Thread/rt-thread #1997: pm组件进入sleep前中断处于关闭状态

geniusgogo, 2018-11-22:
其实有看过我提交的几个低功耗bsp就知道我在处理sleep时，有特殊处理中断开关。
因为并不是所有的处理器都支持WFE唤醒，有的处理器只支持WFI，那么disable掉中断后就无法通过中断唤醒。映像中是MKL15系列就是这样的。

aozima, 2018-11-22:
关注

geniusgogo, 2018-11-22:
第二个问题，退出sleep后立马打开了中断，如果这是被中断唤醒的，那么会立刻跳转到中断去执行，此时os的tick没有校准的，巨坑！！！

HubertXie, 2018-11-22:
现在PM组件还没有哪个BSP使用，没测试过效果。

BernardXiong, 2018-11-22:
@geniusgogo 可以测试，然后PR补丁

geniusgogo, 2018-11-23:
@BernardXiong 先讨论问题，显而易见的推理还是需要的，这些问题是我遇到过的。
那么第三个问题又来了，退出sleep后os tick校准的问题。
`if (delta_tick)
                {
                    /* adjust OS tick */
                    rt_tick_set(rt_tick_get() + delta_tick);
                    /* check system timer */
                    rt_timer_check();
                }
此时没有对当前任务的tick进行校准，所以导致的问题是当前任务tick只要没有tick中断产生就再也得不到：
`-- thread->remaining_tick;
    if (thread->remaining_tick == 0)
    {
        /* change to initialized tick */
        thread->remaining_tick = thread->init_tick;

        /* yield */
        rt_thread_yield();
    }
如此系统就会永远sleep了。`

geniusgogo, 2018-11-23:
第三个问题，idle任务本身在唤醒后没有进行tick校准，这就导致和idle任务想同优先级的其它任务调度出现问题。

HubertXie, 2018-11-23:
借牛头哥的地盘，我也说个问题，现在idle采用的是list，用户可能会注册多个idle_hook，现在都使用组件初始化，这样每个idle_hook调用的顺序没法确定。PM组件是其中一个，这会导致一个问题，PM进入低功耗时，排在后面的hook没被调用，当退出低功耗后，会去执行更高优先级的任务。建议将PM组件的hook固定为最后一个，这样就可以解决这个问题。

enkiller, 2018-11-23:
@HubertXie 排在低功耗后面的 list 会在下一次切回 idle 线程后执行。

HubertXie, 2018-11-23:
有些hook希望在睡眠前处理完，例如看门狗喂狗，下一次切回idle时，看门狗可能就超时了。

geniusgogo, 2018-11-24:
> 借牛头哥的地盘，我也说个问题，现在idle采用的是list，用户可能会注册多个idle_hook，现在都使用组件初始化，这样每个idle_hook调用的顺序没法确定。PM组件是其中一个，这会导致一个问题，PM进入低功耗时，排在后面的hook没被调用，当退出低功耗后，会去执行更高优先级的任务。建议将PM组件的hook固定为最后一个，这样就可以解决这个问题。

这个list确实是串行的，后续估计要考虑hook的优先级？或者说idle中原本就该把sleephook放到最后调用？

geniusgogo, 2018-11-24:
眼下，PM组件使用时需要注意：1.不要创建与idle任务优先级相同的任务。2.arm平台上使用WFE进入sleep。3.唤醒中断处理程序不要做和时间相关的操作（这个应该会后续修改一下PM组件，唤醒后校准完TICK再开中断）。

BernardXiong, 2018-11-24:
这个后面需要把tick补偿放到开中断前。然后可能还需要做一些options

@geniusgogo 天才，提PR吧

closed 2019-02-22 by geniusgogo