RT-Thread/rt-thread #2060: macOS 下 gcc version 7.3.1 编译过不了

erfengwelink, 2018-12-11:
# 1）rtt的libc库libc_signal.h中的“union sigval”、“struct sigevent”、“siginfo_t”和arm-none-eabi-gcc 内部变量命名冲突。

➜  qemu-vexpress-gemini git:(master) ✗ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/application.o
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/libc/libc_signal.h:67:0,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtlibc.h:21,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtdef.h:1070,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtthread.h:24,
                 from applications/application.c:15:
/usr/local/Caskroom/gcc-arm-embedded/7-2018-q2-update/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/sys/signal.h:42:7: error: redefinition of 'union sigval'
 union sigval {
       ^~~~~~
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtlibc.h:21:0,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtdef.h:1070,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtthread.h:24,
                 from applications/application.c:15:
/Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/libc/libc_signal.h:27:7: note: originally defined here
 union sigval
       ^~~~~~
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/libc/libc_signal.h:67:0,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtlibc.h:21,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtdef.h:1070,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtthread.h:24,
                 from applications/application.c:15:
/usr/local/Caskroom/gcc-arm-embedded/7-2018-q2-update/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/sys/signal.h:47:8: error: redefinition of 'struct sigevent'
 struct sigevent {
        ^~~~~~~~
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtlibc.h:21:0,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtdef.h:1070,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtthread.h:24,
                 from applications/application.c:15:
/Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/libc/libc_signal.h:35:8: note: originally defined here
 struct sigevent
        ^~~~~~~~
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/libc/libc_signal.h:67:0,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtlibc.h:21,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtdef.h:1070,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtthread.h:24,
                 from applications/application.c:15:
/usr/local/Caskroom/gcc-arm-embedded/7-2018-q2-update/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/sys/signal.h:72:3: error: conflicting types for 'siginfo_t'
 } siginfo_t;
   ^~~~~~~~~
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtlibc.h:21:0,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtdef.h:1070,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/rtthread.h:24,
                 from applications/application.c:15:
/Users/ouhomarebon/esp_project/RT-Thread/rtt-11-12-2018/rt-thread/include/libc/libc_signal.h:54:24: note: previous declaration of 'siginfo_t' was here
 typedef struct siginfo siginfo_t;
                        ^~~~~~~~~
scons: *** [build/applications/application.o] Error 1
scons: building terminated because of errors.

使用编译器版本如下：
gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (GNU Tools for Arm Embedded Processors 7-2018-q2-update)

# 2）人为修改libc库下的命名后，依然有类型找不到的错误（查看已包含header）：

➜   qemu-vexpress-gemini git:(master)  ✗ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/lcd_init.o
In file included from /Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/components/net/sal_socket/include/dfs_net/sys_select/sys/select.h:14:0,
                 from /usr/local/Caskroom/gcc-arm-embedded/7-2018-q2-update/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/sys/types.h:68,
                 from /usr/local/Caskroom/gcc-arm-embedded/7-2018-q2-update/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/sys/signal.h:12,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/include/libc/libc_signal.h:67,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/include/rtdef.h:461,
                 from /Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/include/rtthread.h:24,
                 from applications/lcd_init.c:1:
/Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/components/dfs/include/dfs_select.h:20:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                      ^~~~~~
/Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/components/dfs/include/dfs_select.h:20:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                       ^~~~~~
/Users/ouhomarebon/esp_project/RT-Thread/rtt-19-3-2018/rt-thread/components/dfs/include/dfs_select.h:20:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                                         ^~~~~~
scons: *** [build/applications/lcd_init.o] Error 1
scons: building terminated because of errors.

基于个人对rtt还不够熟悉，希望rtt team能fix it，谢谢🙏！

BernardXiong, 2018-12-12:
请删除cconfig.h，用目前github上的版本应该可以编译成功。

如果有pthread相关的冲突，请先行关闭POSIX Thread选项。

erfengwelink, 2018-12-13:
使用熊大给的gcc链接https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads 编译可以通过 即使是最新版本

closed 2018-12-18 by BernardXiong

jundazhu, 2019-01-26:
GCC 4.9.3 的工具链可以编译通过，包括pthread相关的都没有冲突。

> 使用熊大给的gcc链接https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads 编译可以通过 即使是最新版本