RT-Thread/rt-thread #2113: list_thread临界区过长

zhaohengbo, 2018-12-24:
finsh的list_thread为了保护object的队列内容，使用了关中断的方式，但是，接下来_list_thread会调用rt_printf打印输出大量数据，由于串口打印速度很慢，因此临界区将变得非常非常长【甚至达到2s】，对于工业系统等高实时系统而言，这是致命的。
我觉得有两种解决方案，但是我觉得都不是特别完美：
1.使用enter_critical方式保护临界区，并使用带有TX缓冲输出的串口驱动【但是此时如果TX缓冲有限，此时相当于依然锁死了线程调度】
2.在临界区拷贝出一份副本，并退出临界区打印【但是对于线程数目过多情况下会消耗大量内存】

BernardXiong, 2018-12-24:
这个地方是必须得考虑的，否则确实没法进入到实际应用中。

@jesven 😢

geniusgogo, 2018-12-25:
伪命题，产品中谁没事执行list_thread?
甚至我都怀疑有多少人（除了睿赛德）会在产品中启用finsh组件？
另一个，不说100%的人吧，99.9%的人执行list_thread只会发生在debug阶段。
so，问题不大。顶多是defect，而不是bug.

heyuanjie87, 2018-12-25:
牛头哥你这是激将法吗

zhaohengbo, 2018-12-25:
> 伪命题，产品中谁没事执行list_thread?
> 甚至我都怀疑有多少人（除了睿赛德）会在产品中启用finsh组件？
> 另一个，不说100%的人吧，99.9%的人执行list_thread只会发生在debug阶段。
> so，问题不大。顶多是defect，而不是bug.
> 其实这个问题就和#2023类似的。

> 伪命题，产品中谁没事执行list_thread?
> 甚至我都怀疑有多少人（除了睿赛德）会在产品中启用finsh组件？
> 另一个，不说100%的人吧，99.9%的人执行list_thread只会发生在debug阶段。
> so，问题不大。顶多是defect，而不是bug.
> 其实这个问题就和#2023类似的。

其实这个就是在一个实际的控制系统上被发现的，有个远端监控系统调用ps等命令询问控制器状态，结果昨天更新完rtt上演了一出好戏

geniusgogo, 2018-12-25:
so.演戏的还真的是那0.1%呢？还是职业的！

heyuanjie87, 2018-12-25:
牛头哥 天使的海洋的产品就被要求要开shell

geniusgogo, 2018-12-25:
我是观众，杀猪哥……

geniusgogo, 2018-12-28:
不想跟你们争论开不开shell。当年预留shell（uart）口的过防爆认证，拿到认证机构那边没一个能过。全部死在uart乱码接收上。所以我们当年在这一块做了很多工作的。是不是那0.1%作为中立商你们自己评估。

今天我就这个BUG的commit来探讨一下目前的修复手段：
鄙人眼拙，看了list_thread的更新log，有如下拙见：
前因：以前list_thread是没有开关中断的。后来简单粗暴的开关中断。所以导致开关时间过长的缺陷。
后果：另，目前的修复可以说是修的火急火燎，我能假设这个修复者针对这个修复没有完全吃透缺陷的上下文。
现在我来摆出鄙人的拙见，如果需要辩论的可以线下加我V信：105884811
为什么说火急火燎，从更新的代码幅度上来看是被这个问题给吓住了，这么大的更新篇幅就为了解决几个变量的原子操作问题。
其实仔细分析，整个list_thread需要访问到的RT_Object_Class_Thread的list是一个从实现上就规避了多任务竞争的问题。所以RT_Object_Class_Thread的list是多任务安全的。可以理解为操作RT_Object_Class_Thread的list不需要加保护。因为对于这个list_thread来说只是读取list。读取操作只会发生在外部修改list，导致迭代失效的问题。而此操作如果是添加list，则不存在迭代失效。删除操作才会导致迭代失效，而删除操作只会发生在idle任务里。所以可以不加保护。但是由于我们还需要进一步操作list里的object的field，准确的说是需要访问thread的stack_addr,stack_sp,stack_size,remaining_tick等.
so,只需要处理好访问这几个field的原子操作即可。
那么如何处理这个最为有效？简单粗暴保护复制当前thread的field即可。

请额外的99.9%的人看一下，是否同意鄙人拙见。

BernardXiong, 2018-12-28:
@geniusgogo 赞，读懂了

BernardXiong, 2018-12-28:
Fix it with #2123

jesven, 2018-12-28:
@geniusgogo RT_Object_Class_Thread的list不需要加保护的说法是不正确的，因为现在已经支持了smp,很有可能是第一个cpu在运行idle的删除操作中，而第二个cpu正在进行访问操作，这种当前任务肯定不会和idle竞争的想法是不再成立了。
现在list_thread是对于每个rt_thread结构，在原子操作的保护中进行了复制，出来后访问的是那个复制品，所以不会再有竞争问题。

HubertXie, 2018-12-28:
实际产品用不用shell，个人觉得看产品类型。
我们的产品，属于物联网节点，一般都没人去操作；量产时都是把shell关了，减少资源占用。

geniusgogo, 2018-12-28:
@jesven 懂了，熊熊刚说了，smp中是每个核有个idle任务，所以会有这样的问题。
其实，即使这样，改动手脚依然可以收敛的。操作手法如上我所述差不多的。
那么现在多核的存在又有一个新的疑问，RT_Object_Class_Thread的list始终是一份吧？那么idle任务中remove掉一个node，既有可能导致当前list迭代失效啊？所以可能导致当前这个node被remove掉，后续的list接不上了啊？

jesven, 2018-12-28:
@geniusgogo 你说的是对的，如果move掉一个，那有可能导致这次迭代中断，后面的显示不出。
但现在由于每次去取的任务结点不是一个,而是批量的，在迭代中会去判断取出的每个结点是否仍有效，有效才会cpy出来显示, 而这种取next的机制，只有在批量的边界那个结点被删，才会导致显示中断。
不过就算是显示中断，敢只是这一次显示不全，当再次list_thread时，又可以去重新打印，反正就是一次显示嘛。
^0^

geniusgogo, 2018-12-29:
@jesven 对，这个就是纠结的地方，显示不全。解决这个问题的本质就是如何处理好关闭中断的时间足够段并且输出完整。如果你用墨菲定律来看待这个问题，鄙人无法接受。o(╯□╰)o

jesven, 2018-12-29:
@geniusgogo 如果片面追求每次的精确显示，那只有把所有信息全部在关中断的情况下cpy出来再打印，但这样的话，就把问题转化成这段内存的来源问题，由于此内存的大小不可预知，且调用此函数时的环境是不定的，很多情况是不能使用malloc, 完全使用栈就更不现实。其实显示不全只是这次不精确而已。

geniusgogo, 2018-12-29:
纵观前后讨论内容：
一会儿说要纳入实际应用中，一会儿99.9%的人都会用，结果来了个概率性出错的修复方案。
那谈啥实际应用，如何面对99.9%的用户呢？
如果list_thread的输出结果原本就不重要就不要这么大费周章的改了吧？
男巫-阿迷-脱发！
撒哈拉-阿西吧-阿门！

jesven, 2018-12-29:
@geniusgogo 不是同一个结果，现在是概率性出错（显示不全），之前有可能是概率性死机，这还是不一样的。起码用户在list_thread时死机了，是更不可接受。

geniusgogo, 2018-12-29:
你看list_thread的修复版本，看最初实现，然后再是加的开关中断，然后再是最近修复方案。你看最初实现。就懂我为什么强调没必要这么改，只需要恢复到最初那种不加开关中断的版本稍微处理一下就可以了。

jesven, 2018-12-29:
请问如果不加关中断应该怎么改呢

geniusgogo, 2018-12-29:
不加中断就不会因为关闭中断太长导致死了吧？
结果当然还是打印不全啊。但是改动最小啊！
`static long _list_thread(struct rt_list_node *list)
{
    int maxlen;
    rt_uint8_t *ptr;
    struct rt_thread *thread;
    struct rt_list_node *node;
    const char *item_title = "thread";

    maxlen = object_name_maxlen(item_title, list);

    rt_kprintf("%-*.s pri  status      sp     stack size max used left tick  error\n", maxlen, item_title); object_split(maxlen);
    rt_kprintf(     " ---  ------- ---------- ----------  ------  ---------- ---\n");
    for (node = list->next; node != list; node = node->next)
    {
        rt_uint8_t stat;
        thread = rt_list_entry(node, struct rt_thread, list);
        rt_kprintf("%-*.*s %3d ", maxlen, RT_NAME_MAX, thread->name, thread->current_priority);

        stat = (thread->stat & RT_THREAD_STAT_MASK);
        if (stat == RT_THREAD_READY)        rt_kprintf(" ready  ");
        else if (stat == RT_THREAD_SUSPEND) rt_kprintf(" suspend");
        else if (stat == RT_THREAD_INIT)    rt_kprintf(" init   ");
        else if (stat == RT_THREAD_CLOSE)   rt_kprintf(" close  ");

#if defined(ARCH_CPU_STACK_GROWS_UPWARD)
        ptr = (rt_uint8_t *)thread->stack_addr + thread->stack_size;
        while (*ptr == '#')ptr --;

        rt_kprintf(" 0x%08x 0x%08x    %02d%%   0x%08x %03d\n",
                   ((rt_uint32_t)thread->sp - (rt_uint32_t)thread->stack_addr),
                   thread->stack_size,
                   ((rt_uint32_t)ptr - (rt_uint32_t)thread->stack_addr) * 100 / thread->stack_size,
                   thread->remaining_tick,
                   thread->error);
#else
        ptr = (rt_uint8_t *)thread->stack_addr;
        while (*ptr == '#')ptr ++;

        rt_kprintf(" 0x%08x 0x%08x    %02d%%   0x%08x %03d\n",
                   (thread->stack_size + (rt_uint32_t)thread->stack_addr - (rt_uint32_t)thread->sp),
                   thread->stack_size,
                   (thread->stack_size + (rt_uint32_t)thread->stack_addr - (rt_uint32_t) ptr) * 100
                        / thread->stack_size,
                   thread->remaining_tick,
                   thread->error);
#endif
    }

    return 0;
}

long list_thread(void)
{
    struct rt_object_information *info;

    info = rt_object_get_information(RT_Object_Class_Thread);
    return _list_thread(&info->object_list);
}`

geniusgogo, 2018-12-29:
上面是最原始的

jesven, 2018-12-29:
双回到原来的问题了，如果在循环访问过程中当前这个结节被删呢，那
for (node = list->next; node != list; node = node->next)
这一句就可能导致死机

geniusgogo, 2018-12-29:
所以我不是前面不是说了只要复制出当前node就行了。

jesven, 2018-12-29:
呵，复制的过程，另一个cpu就可能正在改它，你有可能复制出一个不对的node哟

geniusgogo, 2018-12-29:
复制前关中断啊，保证复制的原子性就行啦

jesven, 2018-12-29:
那现在不就是这样做的么，不同的只是一个和批量的差别哦 XD

geniusgogo, 2018-12-29:
差别大啊，代码改动大啊！

geniusgogo, 2018-12-29:
一个是重构一样的改动能一样吗？

geniusgogo, 2018-12-29:
嗯

geniusgogo, 2018-12-29:
@jesven 看一下我的修改是不是给人感觉跟和谐？
`list_thread()
{
    for (node = list->next; node != list; node = node->next)
    {
        disable()

        if (!is_thread_object_class(node))
        {
            
            break
        }

        backup = *node
        node = node->next

        enable()
        
        proc(backup)
    }
}`

chenyingchun0312, 2019-01-11:
我就是来看看大佬们激烈的讨论的，哈哈，什么时候我也有资格和大佬们一起讨论啊，目前还在学习rtt