RT-Thread/rt-thread #2118: Failed to build qemu-vexpress-a9 using scons in v3.1.1

djndl1, 2018-12-25:
System: Ubuntu 18.04, python 2.7.15, scons v3.0.1
I tried to build this bsp using scons, it gave the following error

> scons: Reading SConscript files ...
AttributeError: 'bool' object has no attribute 'split':
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/bsp/qemu-vexpress-a9/SConstruct", line 27:
    objs = PrepareBuilding(env, RTT_ROOT, has_libcpu=True)
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/tools/building.py", line 382:
    objs = SConscript('SConscript', variant_dir=bsp_vdir, duplicate=0)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 614:
    return method(*args, **kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 551:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 256:
    call_stack[-1].globals)
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/bsp/qemu-vexpress-a9/SConscript", line 12:
    objs = objs + SConscript(os.path.join(d, 'SConscript'))
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 614:
    return method(*args, **kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 551:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 256:
    call_stack[-1].globals)
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/bsp/qemu-vexpress-a9/packages/SConscript", line 10:
    objs = objs + SConscript(os.path.join(item, 'SConscript'))
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 614:
    return method(*args, **kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 551:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 256:
    call_stack[-1].globals)
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/bsp/qemu-vexpress-a9/packages/micropython-latest/SConscript", line 26:
    group = DefineGroup('MicroPython', src, depend = ['PKG_USING_MICROPYTHON'], CPPPATH = path, LOCAL_CCFLAGS = LOCAL_CCFLAGS)
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/tools/building.py", line 604:
    group['LOCAL_CCFLAGS'] = utils.GCCC99Patch(group['LOCAL_CCFLAGS'])
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/tools/utils.py", line 275:
    if VersionCmp(gcc_version, "4.8.0"):
  File "/home/djn/Documents/GitHub/rt-thread-3.1.1/tools/utils.py", line 248:
    la = ver1.split('.')

I built qemu-vexpress-a9 in v3.1.0 and qemu-vexpress-gemini in v3.1.1 sucessfully. At first I thought this might be caused by micropython, after removing micropython, it still happened. It seems that this is somehow related to the newly added "remove -std=99" stuff. Dunno what went wrong,  did I miss something?

BernardXiong, 2018-12-25:
Hi,
what's your gcc version? it failed on gcc version detection.

djndl1, 2018-12-25:
Problem solved after switching to gcc 6.

closed 2018-12-25 by djndl1