RT-Thread/rt-thread #2138: [bsp][stm32][stm32f429-atk-apollo]缺少"pcf8574.h"文件

fsyxhua, 2018-12-29:
[bsp][stm32][stm32f429-atk-apollo]缺少"pcf8574.h"文件

balanceTWK, 2018-12-29:
[image]

没有复现你的问题，你是不是在输入命令 scons --target=xxx 前 没有输入命令 pkgs --update ？

fsyxhua, 2018-12-30:
pkgs --update 肯定输入了，而且我都设置是自动update的，手动输入也试了，能帮我看看pcf8574.c和pcf8574.h的文件路径吗？我的直接搜索都搜索不到的

fsyxhua, 2018-12-30:
谢谢balanceTWK的回答，问题找到了，从gitee上clone下来的代码没有drivers文件夹，刚才去gitee站点看了一下，目录里确实没有drivers文件夹
[image]
[image]

fsyxhua, 2018-12-30:
menuconfig 配置里面没有，pcf8574的选项，需要手动拷贝进去，你那个是不是手动添加的？

closed 2018-12-30 by fsyxhua

balanceTWK, 2018-12-30:
@fsyxhua  没有手动加。代码是 GitHub 最新的， 选中 以太网 选项，按照正常流程打开 就能用了。