RT-Thread/rt-thread #2162: lwip处理ip包 出现data abort

Sparks-go, 2019-01-07:
zynq7000平台，芯片zynq7020移植lwip2.0.2 rt-thread运行时，协议栈
err_t
ip4_input(struct pbuf *p, struct netif *inp)
{
*******************************************
*******************************************
 /* copy IP addresses to aligned ip_addr_t */
  ip_addr_copy_from_ip4(ip_data.current_iphdr_dest, iphdr->dest);
  ip_addr_copy_from_ip4(ip_data.current_iphdr_src, iphdr->src);
*********************************************
*********************************************
}拷贝地址时出现 data abort
ip_addr_copy_from_ip4，当访问iphdr的结构体变量dest和src时，出现data abort
验证了iphdr 地址没有问题，访问其他结构体变量都可以正常访问，但是只要访问dst src 就会异常

BernardXiong, 2019-01-19:
是否是数据未对齐的缘故？

mysterywolf, 2022-03-05:
高度怀疑是memcpy对齐的问题，现在已经改成rt_memcpy了

closed 2022-03-05 by mysterywolf