RT-Thread/rt-thread #2272: [sfud]retry

PiscesLyn, 2019-01-26:
` /* 100 microsecond delay */`
   ` flash->retry.delay = retry_delay_100us;`
    `/* 60 seconds timeout */`
   ` flash->retry.times = 60 * 10000;`

`static void retry_delay_100us(void) {`
 `   /* 100 microsecond delay */`
 `   rt_thread_delay((RT_TICK_PER_SECOND * 1 + 9999) / 10000);`
`}`
retry_delay_100us通过rt_thread_delay实现，但rt_thread_delay取决于系统clock颗粒度，目前最低只能1ms，比如RT_TICK_PER_SECOND值为1000，这边实际导致的delay就是1ms而不是100us。
底下times按照100us去计算 的count，实际结果变成了600s。其实连60s都嫌多，在设备初始化过程的retry个人觉得1s就够了.

armink, 2019-01-26:
这些属性用户也都是可以通过 flash device 对象来二次修改。

毕竟 RTT 缺少 busy_wait 机制，所以暂时使用 thread_delay ，如果你的 tick 粒度低于 1000，SFUD 是有警告提示用户：“耗时可能过多”

PiscesLyn, 2019-01-26:
ok,了解,谢谢解答.

closed 2019-01-26 by PiscesLyn