RT-Thread/rt-thread #2307: 不同编译器 rt_int32_t 定义问题

xuzhuoyi, 2019-02-02:
https://github.com/RT-Thread/rt-thread/blob/903d1d73ca6637fa5e98d26bb88ddea92972246a/include/rtdef.h#L60-L74
这段代码中的
`typedef signed   int                    rt_int32_t;     /**< 32bit integer type */`
能不能改回原来的`typedef signed long rt_int32_t`
因为有一些 32 位 DSP 和 8/16 位单片机的编译器将 int 视为 16 位长度。 
或许可以改为如下：
```
typedef signed   char                   rt_int8_t;      /**<  8bit integer type */
typedef signed   short                  rt_int16_t;     /**< 16bit integer type */
typedef unsigned char                   rt_uint8_t;     /**<  8bit unsigned integer type */
typedef unsigned short                  rt_uint16_t;    /**< 16bit unsigned integer type */

#ifdef ARCH_CPU_64BIT
typedef signed int                        rt_int32_t;     /**< 32bit integer type */
typedef signed long                     rt_int64_t;     /**< 64bit integer type */
typedef unsigned int                    rt_uint32_t;    /**< 32bit unsigned integer type */
typedef unsigned long                   rt_uint64_t;    /**< 64bit unsigned integer type */
#else
typedef signed long                        rt_int32_t;     /**< 32bit integer type */
typedef signed long long                rt_int64_t;     /**< 64bit integer type */
typedef unsigned long                    rt_uint32_t;    /**< 32bit unsigned integer type */
typedef unsigned long long              rt_uint64_t;    /**< 64bit unsigned integer type */
#endif
```

xuzhuoyi, 2019-02-02:
https://github.com/RT-Thread/rt-thread/blob/903d1d73ca6637fa5e98d26bb88ddea92972246a/src/kservice.c#L1268-L1292
同样，这里 __rt_ffs() 函数参数为 int 类型，在移植到 dsp 的时候发现实际只有 16 位，线程优先级多的时候会出错，能否也改为 rt_int32_t？

xuzhuoyi, 2019-02-02:
>           可以考虑使用int16_t/uint16_t这种具有确定长度数据类型的C99标准定义

但是现在问题是 rtdef.h 定义的 32 位类型 rt_int32_t 在 DSP 的编译器上实际是 16 位，而 RT-Thread 代码大量使用了 rt_int32_t 类型，所以造成了一些问题，所以最好修改 rtdef.h 的定义。

liruncong, 2019-02-02:
把int定义为16位的比较少，可以自己修改下rtdef.h文件定义，类似如下：
typedef int8_t                          rt_int8_t;      /**<  8bit integer type */
typedef int16_t                         rt_int16_t;     /**< 16bit integer type */
typedef int32_t                         rt_int32_t;     /**< 32bit integer type */
typedef uint8_t                         rt_uint8_t;     /**<  8bit unsigned integer type */
typedef uint16_t                        rt_uint16_t;    /**< 16bit unsigned integer type */
typedef uint32_t                        rt_uint32_t;    /**< 32bit unsigned integer type */

liruncong, 2019-02-02:
内核中还有很多直接使用int的，这个好像没什么好办法，自己改吧

BernardXiong, 2019-02-03:
这个好像有些怪异了。DSP是16位的吗？

BernardXiong, 2019-02-03:
基本上8位，16位芯片是不在考虑之内。或许可以对rtdef.h中区分编译器进行定义

mysterywolf, 2021-02-24:
目前，RTT仅针对32或64

closed 2021-02-24 by mysterywolf