RT-Thread/rt-thread #2317: RT-Thread English Forum

WGR7, 2019-02-10:
Hi all,

Is there still exists the RTT english forum?

I have some doubts regarding task management.  Is the only way to restart a task is delete and create it again? What is the diffetence between detach and delete?

Thanks

majianjia, 2019-02-10:
Hi,
To answer your question, "detach" is for the instance who was "init". While "delete" is for who was "create".

They must be used in pairs. thread_init - with - thread_detach, thread_create - thread_delete.

The difference is the former initiates a thread handle from static mem, such as global variables or static variable, the later creates handle by dynamic mem allocating, and it will return the pointer to the thread handle, which must be freed by "delete" to avoid mem leaks.

In your case, use init-detach should be better. 

About Eng. forums... as far as i know they are working on it.

Hope it helps
Jianjia

BernardXiong, 2019-02-12:
Thank @majianjia , you are right!

WGR7, 2019-02-12:
Thanks for your support.

I have another question. How the parameter timeslice works? If a have three tasks with same priority and a timeslice of 20, how the execution will happens?

What determine the execution order of three tasks with the same priority? Will be the creation order?

Best regards

majianjia, 2019-02-12:
> Thanks for your support.
>
> I have another question. How the parameter timeslice works? If a have three tasks with same priority and a timeslice of 20, how the execution will happens?
>
> What determine [...]

If all 3 tasks are running in a big while(1) loops and never release the CPU (could be released by waiting for events, smartphones... or calling rt_thread_delay(), rt_thread_yield()). 
In this case, they will be called one-by-one by the order in the ready list, each of them can run 20 ticks.
This list could be in the creation order, but it depends. I think it performed like a First-In-First-Out. 

@BernardXiong 
I am not sure if one of them call rt_thread_yield(), will it return to the current task or the next?

BernardXiong, 2019-02-13:
For same priority level thread, 

In each OS tick ISR, the remaining_tick of current thread will be subtracted by one. When remaining_tick is 0, current thread will do rt_thread_yield(), that is to release CPU and be put to the end of this priority ready queue.

WGR7, 2019-02-13:
So, if each thread has a timeslice of 5, then

Thread0  -> Thread1 -> Thread2
  5 ticks  -> 5 ticks  -> 5 ticks

So, each thread will be in idle for 10 ticks until it will be ready again?

majianjia, 2019-02-13:
as mentioned by BernardXiong, yes, when a task run out of its timeslice, it will yield itself, and put itself to the end of ready list.

if currently thread 0 run out 
Thread0 -> Thread1 -> Thread2
0 ticks -> 5 ticks -> 5 ticks

then the ready list will become this
Thread1 -> Thread2 -> Thread0
5 ticks -> 5 ticks -> 5 ticks

WGR7, 2019-02-15:
Excellent.

Could you tell me if I call rt_interrupt_enter function, if any interrupt occurs until I call rt_interrupt_exit, will be serviced after rt_interrupt_exit?

majianjia, 2019-02-15:
yes, interrupt will be pending untill it is enabled. same as other rtos

BernardXiong, 2019-02-23:
In the future, GitHub issue is mainly used for communication in English.

dosadi, 2019-03-26:
Hi,

I am playing with the current RT-Thread tree, trying to get bsp/simulation to work.  I have run into an RT_ASSERT() in the core kernel code:

(gdb) bt
#0  0x0000555555575362 in rt_assert_handler (ex_string=0x5555555813a8 "rt_object_get_type((rt_object_t)thread) == RT_Object_Class_Thread", 
    func=0x555555581530 <__FUNCTION__.3194> "rt_thread_startup", line=291) at /home/jon/rt-thread/src/kservice.c:1340
#1  0x000055555557663e in rt_thread_startup (thread=0x7fffffffdfa0) at /home/jon/rt-thread/src/thread.c:291
#2  0x000055555557cf13 in main () at applications/startup.c:141

Calling code:

    rt_thread_t test_thread_1;
    rt_thread_t test_thread_2;

    /* disable interrupt first */
    rt_hw_interrupt_disable();
    
    /* startup RT-Thread RTOS */
    rtthread_startup();

    
    test_thread_1 = rt_thread_create(
                                        "test thread 1",
                                        test_thread_1_thread,
                                        RT_NULL,
                                        512,
                                        20,
                                        20);
         
    if (test_thread_1 != RT_NULL)
    {
        printf("aa\n");
        rt_thread_startup(&test_thread_1);
        printf("bb\n");
    }

Any ideas?

Jon

mysterywolf, 2021-02-06:
This is rt-thread forum: https://club.rt-thread.io/

closed 2021-02-06 by mysterywolf