RT-Thread/rt-thread #2354: 一个奇怪的问题：串口初始化波特率会影响线程是否起作用

SuWeipeng, 2019-02-19:
遇到一个奇怪的问题：串口初始化的波特率是115200，导致一个与串口无关的线程失效。

# 解决办法
将初始化波特率调大

SuWeipeng, 2019-02-19:
# 不明白为什么会这样，希望得到帮助和解答。
下面我会把试验的代码放上来。

SuWeipeng, 2019-02-19:
代码链接如下，仿真呈现的问题与我用stm32f4discovery开发板呈现的问题一样，解决方法也一样。

[https://pan.baidu.com/s/1hIIHKD9a308rKn_qOmHq7A](https://pan.baidu.com/s/1hIIHKD9a308rKn_qOmHq7A)

SuWeipeng, 2019-02-19:
[image]
[image]

SuWeipeng, 2019-02-19:
# 有问题的代码
## 代码在 User 文件夹下，写得很简单，100% 复现问题

[STM32F103_StdPeriph.zip](https://github.com/RT-Thread/rt-thread/files/2882316/STM32F103_StdPeriph.zip)
[STM32F4Discovery_HAL.zip](https://github.com/RT-Thread/rt-thread/files/2882317/STM32F4Discovery_HAL.zip)

SuWeipeng, 2019-02-19:
@BernardXiong  不知道此问题是什么原因，求助。

balanceTWK, 2019-02-20:
[image]

感觉你魔改了很多啊，没法看😓

SuWeipeng, 2019-02-20:
> [image]
> 
> 感觉你魔改了很多啊，没法看😓
> 
> 为啥不直接使用 官方的 BSP 😶

只有 User 文件夹下的是我写的，别的都是自动生成的，STM32F103的是用的“标准外设库”，驱动(drv_xxx)都是从 RT-Thread 的github项目上考出来的。

自动生成就是重构了一下文件。

SummerGGift, 2019-02-20:
学习的话也请使用 stm32 文件夹下的新 bsp，旧的使用 STD 库的 BSP 往后将逐步不再维护了的。

SuWeipeng, 2019-02-20:
@SummerGGift
这个问题我是用stm32f4discovery开发板发现的，用的是CubeMX生成的HAL库，驱动是考的RT-Thread官方 bsp/stm32 目录下的。

发现这个问题后，想看看是不是跟hal库相关，同时也想仿真看下会不会有同样的现象。所以用标准外设库试了一下。

现在基本确定这个现象跟HAL或者标准外设库无关。

BernardXiong, 2019-02-23:
感觉是不应该的。"线程是否起作用" 这个指的是什么意思？

SuWeipeng, 2019-02-24:
> 感觉是不应该的。"线程是否起作用" 这个指的是什么意思？

编程是用PA0按键的线程挂起PB14的线程，”线程不起作用“是说波特率调小后，PA0按键线程不能挂起PB14。

PA0和PB14是各自独立的线程，与串口一点关系都没有。

closed 2021-02-25 by mysterywolf