RT-Thread/rt-thread #2363: AT组件判断URC和指令返回不够严谨

geniusgogo, 2019-02-22:
受熊委托拜读AT client组件。

现在发现两个坑：
1.处理接收数据urc的时候，有可能回调urc的时候带了一部分数据，而另一部分数据还需要继续接收。
2.+CGREG：这条urc和主动查AT+CGREG?返回码一样。这就有个问题，如果同时注册了这条URC，那么
主动查询将得不到正确返回。

BernardXiong, 2019-02-22:
我委托你个毛线，像是我求你一样么

closed 2019-02-22 by BernardXiong

Lawlieta, 2019-02-22:
1. 我理解你这里的回调 URC 中带的数据是用来解析接收数据的信息的命令，比如 esp8266 中 服务器向模块发送10 bytes 数据格式如下：
```c
+IPD:1,10:abcdefghij
```
这里 URC 回调中需要解析的就是 “+IPD:1,10:” 数据，读取接收的 socket 为 1 和数据的长度为 10，然后再在回调中调用 `at_client_recv` 函数，直接从串口中读取 10 bytes 数据，这是整个接收数据的流程。所以 ESP8266 中 URC 处理函数如下：

```c
static void urc_recv_func(const char *data, rt_size_t size)
{
    /* get the current socket and receive buffer size by receive data */
    sscanf(data, "+IPD,%d,%d:", &socket, &bfsz);
    ...
   (at_client_recv(recv_buf, bfsz, timeout);
   ....
}
static struct at_urc urc_table[] = {
         .....
        {"+IPD",             ":",              urc_recv_func},
};
```
这样只要在 URC 回调函数中处理好数据接收应该不会出现数据丢失的问题。

2. AT+CGREG? 函数一般在初始化的时候调用，一般在 URC 没初始化之前就调用，之后的 +CGREG：应该只会由 URC 处理

geniusgogo, 2019-02-22:
> 我委托你个毛线，像是我求你一样么

嗯，你没求我，你只是鄙视。说什么“连你都不用，这东西还要怎么弄。真心鄙视你”。
原话，OK，没必要有这种情绪。过分去解读文字没有意义。
如果以觉得我这句话说得不好，嗯，我这里声明一下，我有点自作主张了，对不起！
那么也请你收回，那种貌似我不用新功能好像违了你的法似得那种口气对我说过的话。