RT-Thread/rt-thread #2365: 建议增加飞思卡尔智能车竞赛所用芯片（nxp）的bsp

Vincent-VG, 2019-02-25:
飞卡有着很强的影响力，有利于rtt在大学生中的进一步推广。
历届的飞卡比赛都指定使用nxp的芯片（以前固定一款 mk66），不知出于什么考虑或原因，几乎没有使用操作系统的，都是裸机编程，每一个新的开始，几乎都要重新编写驱动。但飞卡又是个重视算法的竞赛，如果使用了rtt，驱动完善，参赛者可以节省更多的时间和精力放在应用上...

PengQianheng, 2019-02-25:
1.2的版本已经提供了K60的BSP

Vincent-VG, 2019-02-25:
没有看到啊， 另外有k66的bsp吗

PengQianheng, 2019-02-25:
分支里有，如下
https://github.com/RT-Thread/rt-thread/tree/stable-v1.2.x/bsp/K60Fxxxx

Vincent-VG, 2019-02-25:
额， 也就是说之后的版本都没这个bsp了

BernardXiong, 2019-02-26:
有这个，

[frdm-k64f](https://github.com/RT-Thread/rt-thread/tree/master/bsp/frdm-k64f)

sogwms, 2019-04-03:
桌睛近期在公众号推过[飞卡 MCU 选型](https://mp.weixin.qq.com/s?__biz=MzA5NjQyNjc2NQ==&mid=2452220908&idx=2&sn=8604f2b5136636984e15dac99ac3dc53&chksm=876d058eb01a8c9899fad1e724ff08dbca491ff3aa955a9a7588724a56e019b4d7e17571cde9&xtrack=1&scene=0&subscene=131&clicktime=1554281659&ascene=7&devicetype=android-28&version=27000334&nettype=3gnet&abtest_cookie=BAABAAoACwASABMABQAklx4AVpkeAM%2BZHgDZmR4A3JkeAAAA&lang=zh_CN&pass_ticket=Fdu5DNANttrnKQVKleDGXwzkDGxk2NgsP96eqBAvQTW9SkbmQw9nAsr8NuNQOqZK&wx_header=1)的文章， 很明显以后 imxrt 系列会是广大参赛者的选择。 建议完善相应的bsp驱动 （PWM驱动，摄像头驱动，显示屏驱动等）

zengwangfa, 2019-06-14:
十分赞同

guidons-master, 2020-04-11:
非常赞同

guidons-master, 2020-04-11:
imxrt系列是大趋势

closed 2022-08-04 by mysterywolf