RT-Thread/rt-thread #2454: sensor驱动编译时找不到头文件

sogwms, 2019-03-14:
```
CC build\packages\spl0601-latest\sensor_goer_spl0601.o
In file included from packages\spl0601-latest\sensor_goer_spl0601.c:1:0:
packages\spl0601-latest\sensor_goer_spl0601.h:4:20: fatal error: sensor.h: No such file or directory
```

如何解决？

BernardXiong, 2019-03-15:
需要使用github上新的代码，sensor相关的是属于近期合并到RT-Thread中的

sogwms, 2019-03-16:
之前已经跟进最新了，   不行的，  刚又重新拉取了下， 还是不行。

sogwms, 2019-03-16:
components/drivers/sensors/sensor.h 是存在的，   是头文件搜索路径不对吗

BernardXiong, 2019-03-17:
@Guozhanxin Please take a look.

Guozhanxin, 2019-03-18:
应该是没有开启sensor驱动框架
[image]

sogwms, 2019-03-18:
是这个问题。   这在文档里都找不到说明啊， 设备驱动，传感器开发指南都没有。  还有这个搞成自动化感觉更好

Guozhanxin, 2019-03-18:
嗯，可以在软件包索引那里加个强依赖（select）

closed 2019-03-20 by sogwms