RT-Thread/rt-thread #2460: finsh无法输入的bug

XXXXzzzz000, 2019-03-15:
现象:
使用finsh时可以输入,有回显,按下回车时不会回显回车,而是从开头继续输入,经debug发现是此处
```
static int finsh_getchar(void)
{
#ifdef RT_USING_POSIX
    return getchar();
#else
    int ch=0;

    RT_ASSERT(shell != RT_NULL);
    while (rt_device_read(shell->device, -1, &ch, 1) != 1)
        rt_sem_take(&shell->rx_sem, RT_WAITING_FOREVER);

    return ch;
#endif
}
```
ch在源码中没有进行初始化,导致莫名其妙的乱码,但是回显是正确的,此处不知道有没有什么更深层次的原因.
并且此处及finsh_thread_entry中仅检查其是否小于0,而我的调试信息显示ch存储的是一个很大的数字.

BernardXiong, 2019-03-15:
是的，这个是一个问题，原因是 #2414 合并导致的，后来由 #2445 修正。你可以再更新代码看看这个问题是否还存在？如果还存在，那么针对这个PR得要重新考虑了。

XXXXzzzz000, 2019-03-15:
ok,我也是刚刚合并以后发现已经修复了.

closed 2019-03-18 by BernardXiong

gbcwbz, 2019-03-19:
直接改成int确实不妥，后面只读取了一个字节，如果CPU是大端的话仍可能出错，还是定义成char在返回之前转为int好些