RT-Thread/rt-thread #2572: 合并 这个之后 程序不运行了 [Kernel] To check object re-initialization in the list

seblee, 2019-04-12:
合并 这个之后 程序不运行了 [Kernel] To check object re-initialization in the list

我仿真了一下

发现一直在这个循环里不出来了

look
```c
/* enter critical */
    rt_enter_critical();
    /* try to find object */
    for (node  = information->object_list.next;
            node != &(information->object_list);
            node  = node->next)
    {
        struct rt_object *obj;

        obj = rt_list_entry(node, struct rt_object, list);
 // return;   
        RT_ASSERT(obj != object);
    }
    /* leave critical */
    rt_exit_critical();
```
我仿真的时候  前面几次都是node  != node->next
后面就都相等了
是不是应该改成
```c
/* enter critical */
    rt_enter_critical();
    /* try to find object */
    for (node  = information->object_list.next;
            node != information->object_list->next);
            node  = node->next)
    {
        struct rt_object *obj;

        obj = rt_list_entry(node, struct rt_object, list);
 // return;   
        RT_ASSERT(obj != object);
    }
    /* leave critical */
    rt_exit_critical();
```
请看截图
[image]

BernardXiong, 2019-04-13:
确认下是否是最新代码。如果这里有问题，应该说明你的代码中有反复初始化对象的问题

BernardXiong, 2019-04-13:
@Lawlieta 这个好像是和SAL相关？testcase中已经有一些和SAL相关的测试用例了吗？如果有，在系统测试中把SAL测试用例都打开吧

Lawlieta, 2019-04-13:
暂时还有针对 SAL的测试用例，但是网络相关的测试用例都使用到 SAL 的。

这里错误情况不好确定是否和 SAL 相关，需要进一步定位到错误的初始化位置 @seblee

seblee, 2019-04-15:
@Lawlieta @BernardXiong  确实重复初始化的问题

closed 2019-04-15 by seblee

armink, 2019-04-15:
重复初始化了，这里没有显示断言吗？

seblee, 2019-04-15:
@armink 是这样的错误 但是没有提示
[image]

BernardXiong, 2019-04-16:
既然前面执行了rt_mutex_create，后面再执行rt_mutex_init，铁定完蛋啊，不是这样使用的。

这样使用，即使是以前的方式，当调用list_mutex命令时，也死循环了。所以现在的方式本身就是要阻止这种错误用法。

seblee, 2019-04-16:
是的  刚接手的时候内核版本是 还是210，最近升级到4了的  前天升级到最新的时候才发现的

这种低级错误，也是简直了，尴了好一尬

armink, 2019-04-16:
好像我们的重复初始化检测，没有检测到用户这个用法错误