RT-Thread/rt-thread #2579: Missing yfuns.h

dosadi, 2019-04-15:
Hi,

In the latest pull (Monday April 15), the file:

yfuns.h

...is #included from:

src/rt-thread/components/libc/compilers/dlib/syscall_read.c

...and is missing from the repository.

BernardXiong, 2019-04-16:
Please provide your IAR version information.

dosadi, 2019-04-16:
What is IAR version information?  I couldn't find anything on a Google
search....

BernardXiong, 2019-04-18:
It's IAR compiler.

These files under dlib folder, are the libc porting for IAR/libc.

dosadi, 2019-04-19:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;} Ok.  I use my own custom build system based on simple makefiles.                                                                                                                                                                        Sent via BlackBerry Hub+ Inbox for Android                                                                                                                                                                           From: notifications@github.comSent: April 18, 2019 11:40 AMTo: rt-thread@noreply.github.comReply-to: reply@reply.github.comCc: dosadi82@gmail.com; author@noreply.github.comSubject: Re: [RT-Thread/rt-thread] Missing yfuns.h (#2579)  It's IAR compiler.
These files under dlib folder, are the libc porting for IAR/libc.

closed 2019-04-21 by BernardXiong