RT-Thread/rt-thread #2614: 把Linux的代码移植过来删除掉原来的协议信息没有问题么？

diverger, 2019-04-24:
初次接触RTT, 发现component下的一些驱动代码和Linux下的极像(比如SDIO), 但是文件header的协议和版权信息完全不同，一个是GPL，一个是 Apache，没有问题么？感觉怪怪的.........

BernardXiong, 2019-04-25:
是这样的？这个看来要 @weety  check下

diverger, 2019-04-25:
RTT和Linux在某些方面设计得和Linux相似，许多驱动稍加修改就可以从Linux移过来，但是Linux是GPL。RTT应该不会局限于国内吧，这个版权问题还是要重视下。可以参考下同为Apache-2.0的https://github.com/zephyrproject-rtos/zephyr

BernardXiong, 2019-04-25:
RT-Thread是一个社区性的操作系统，一般来说，能够合并的代码会严密审核。在这个过程中可能也难免会遇到些问题，所以如果有这方面的可能，还请一一指出 🙏

diverger, 2019-04-25:
这些开源的版权兼容问题很麻烦，可以在开始的时候就注意一下，靠近Linux有好处，驱动移植方便，但是太方便了会影响创新，别成了另一个Linux。还是希望咱们国产的RTOS能越来越好。

BernardXiong, 2019-04-26:
“靠近Linux” 仅是一些风格上的吧，类似UNIX命名风格、标准向POSIX靠近等。至于说Linux驱动很容易就可以移植过来，希望是美好的，现实还是挺骨感的。

RT-Thread不是另外一个Linux，这个是肯定的。所以我也不会去看Linux的内核代码，对Linux内核也不甚了解呐。对于RT-Thread还是坚持“小而美”的特点

diverger, 2019-04-26:
嗯，我主要是指的驱动。有网络宣传RTT要"把Linux纳入进来”，是这样的吗？这个“纳入”是怎么纳入呢？

BernardXiong, 2019-04-26:
“把Linux纳入进来” 这个是没有的事，也是不可能的啊。

还是说把Linux下也能够开发RTT，也纳入到考虑中？

diverger, 2019-04-26:
:), 那看来是网络谣传了。

closed 2021-02-25 by mysterywolf