RT-Thread/rt-thread #2631: scons创建mdk工程时，group组中有多个library的情况下，会重复创建同名group

ericQiang, 2019-04-29:
条件：
在bsp工程目录下，增加libs文件夹，内有多个library库文件，现要通过scons --target=mdk5将这几个library文件添加到名为Ext_libs的group中。
libs目录下的SconScript脚本内容如下：
`import rtconfig
from building import *

cwd     = GetCurrentDir()
src = []
path = ['']
path += [cwd]
libs = ['']
if rtconfig.CROSS_TOOL == 'keil':
    libs += ['xx_lib_v20']
    libs += ['xx_lib20']

libpath = ['']
libpath += [cwd]
group = DefineGroup('Ext_libs', src, depend = [''], CPPPATH = path, LIBS = libs, LIBPATH = libpath)

Return('group')`
最终scons --target=mdk5生成的工程中包含两个名为Ext_libs的group。两个group下分别添加了xx_lib_v20.lib和xx_lib20.lib。

针对此问题，我修改Tools目录下的keil.py，并提交pull request，但好像自动测试没有通过（不过我在bsp下做过测试是OK的）。由于这是我首次尝试代码贡献，所以不知后续如何处理。请各位大佬协助一下。谢谢。

BernardXiong, 2019-04-29:
这个和自动测试关系不大，主要是你的github账号和git中配置的账号有问题，不匹配，导致CLA错误

ericQiang, 2019-04-29:
哦。好的。谢谢熊大。

BernardXiong, 2019-04-30:
merge with #2635 , thank you for your contribution🥇

closed 2019-04-30 by BernardXiong