RT-Thread/rt-thread #2646: 建议增加 NXP LPC3250 处理的 BSP

terrychx, 2019-05-04:
建议增加LPC3250 处理器的  BSP

BernardXiong, 2019-05-04:
这个应该是挺老的ARM9吧，有需求的话欢迎贡献，欢迎PR

terrychx, 2019-05-05:
我们产品中一直用这个芯片，想移植一个RTOS，正好想用国产的RT Thread。

 

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Bernard Xiong
发送时间: 2019年5月4日 19:03
收件人: RT-Thread/rt-thread
抄送: terrychx; Author
主题: Re: [RT-Thread/rt-thread] 建议增加 NXP LPC3250 处理的 BSP (#2646)

 

这个应该是挺老的ARM9吧，有需求的话欢迎贡献，欢迎PR

yygg, 2019-05-05:
我回去搜一下老库存 我好像移植过LPC3250 用的是周立功的那个开发板

terrychx, 2019-05-05:
谢谢！方便的话，我可以帮助做一部分工作，我的IDE环境是KEIL4。

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Urey
发送时间: 2019年5月5日 9:39
收件人: RT-Thread/rt-thread
抄送: terrychx; Author
主题: Re: [RT-Thread/rt-thread] 建议增加 NXP LPC3250 处理的 BSP (#2646)

 

我回去搜一下老库存 我好像移植过LPC3250 用的是周立功的那个开发板

closed 2021-05-16 by mysterywolf