RT-Thread/rt-thread #2663: stm32f4xx-HAL 编译 IWG代码错误

TheTravels, 2019-05-09:
在 master/bsp/stm32f4xx-HAL/drivers/drv_iwg.c源文件的 drv_control函数有如下代码：
`
static rt_err_t drv_control(rt_watchdog_t *wdt, int cmd, void *arg)

{

    switch (cmd)

    {

    case RT_DEVICE_CTRL_WDT_SET_TIMEOUT:

        hiwdg.Init.Reload = (rt_uint32_t)*arg;

        if (HAL_IWDG_Init(&hiwdg) != HAL_OK)

        {

            return RT_ERROR;

        }

        break;
`
其中 " (rt_uint32_t)*arg;"这个强制类型转换是错误的为什么就没人发现？？？
我用的 MDK5编译错误信息为：“drivers\drv_iwg.c(35): error:  #171: invalid type conversion”，经检查发现 arg是 void指针，必须转换类型才能取值。我没有使用其它编译器测试，但是根据我看到的制作 BSP的文档，发布 BSP包是需要对gcc，iar，mdk进行测试的，所以为什么还会出现对 void指针取值的这种错误？？？唯一合理的解释就是官方在制作 BSP的时候本身并没有遵循严格的测试，而恰好官方所使用的编译器在对 void指针取值的时候将 “void *”与某一类型视作相同如 “char*”，而根据语句本身可以判定编译器并没有视作”uint32_t*“。当然我在使用过程中并不只是发现这一个错误，在我添加 C++组件的时候会因为符号 ”SHT$$INIT_ARRAY$$Base“未定义而无法编译。我知道这是用来调用 C++构造函数的，我也测试过使用 new动态创建对象即使 cplusplus_system_init函数未执行对程序也没有影响，因为这里调用的是静态构造函数，问题是如果有谁在使用过程中创建了一个静态对象，而由于上述符号未定义便直接将函数注释成空，那么程序在运行的时候问题就来了，如果你在构造函数中做了一些初始化显然是会失败的。虽然这些问题不是不能解决的，但是试想一下，假如你在信任 RT-Thread的前提下你肯定是配置好内核直接用的，你肯定是专注与自己的代码，而这个时候你突然发现某个功能不正常了你是很难找到问题的，因为对 RT-Thread的信任导致你不会怀疑 RT-Thread会有 bug。
说这么多，一个是希望修复这两个问题，另一个是希望官方在发布 BSP的时候自己能严格遵守 BSP发布的要求进行充分的测试，如果你们自己在制作 BSP的时候都不遵守你还能指望别人在上传 BSP的时候严格遵守吗？？？？

BernardXiong, 2019-05-09:
好像说得挺对的。后面可以看看rt-thread/bsp/stm32下是否也有这类的问题。

而这个stm32f4xx-HAL bsp，后期是考虑删除掉，是老的bsp了。

TheTravels, 2019-05-09:
rt-thread/bsp/stm32下与我使用的 IC相关的我会自己做些测试，我是初次使用 rt-thread，对其中很多东西还不是很熟悉，所以就浏览了你们比较多的文档，然后想自己移植一个 BSP，然后又想用 C++。但是我发现许多 BSP都可以配置 C++组件，但是就目前我找到配置之后直接编译通过的只有 F7的BSP包，我测试过的其他几个 F4的BSP以及 LPC的 BSP都是在编译的时候本来应该用 ”arm-none-eabi-g++“编译 C++源文件（命令行直接编译）却变成了用 ”g++“编译，因为对 scons不熟，直到终于找到 F7配置 C++组件编译通过，经对比并更改 F4的配置编译通过了才知道 rtconfig.py和 SConstruct两个文件缺少 CXX配置项。而且我发现最新的代码至少在 F4我目前没有看到有用标准库的，这个也许是我没有全部浏览完，因为同事用的标准库，所以为了尽快熟悉并少出问题我选择基于标准库移植，但是我只能版本退回。而且我在测试的时候还发现有不支持 IAR的。另外就是新版本 BSP全部使用 HAL库，但是 HAL和 LL两个库是兼容的，LL库会被编译，但是我没有看到相关文档说明如何使用 LL库，使用 STM32CubeMX 配置初始化默认使用的是 HAL库，我还没有去深究代码中如何使用 LL库，但从目前的文档来看是不支持的，在某些场合对效率有要求的话 LL库还是有优势的。最后说一点就是模板工程 stm32f4xx-HAL对 STM32CubeMX 并不支持，不知你们是基于什么原因的考虑？

SummerGGift, 2019-05-09:
> rt-thread/bsp/stm32下与我使用的 IC相关的我会自己做些测试，我是初次使用 rt-thread，对其中很多东西还不是很熟悉，所以就浏览了你们比较多的文档，然后想自己移植一个 BSP，然后又想用 C++。但是我发现许多 BSP都可以配置 C++组件，但是就目前我找到配置之后直接编译通过的只有 F7的BSP包，我测试过的其他几个 F4的BSP以及 LPC的 [...]

stm32f4xx-HAL 是比较老的 BSP 了，并不支持 STM32 cubemx 配置，他也不是模板工程。新的 bsp 都在 stm32 目录下。

TheTravels, 2019-05-09:
我以为是模板工程主要两点，一 BSP命名”f4xx“，通常的理解就是 F4系列，二，menuconfig中有 ” Device type (STM32F407VG)  --->“可以配置芯片型号。而在 ”bsp/stm32“目录下都是基于具体开发板的也不存在模板工程，当然这个我可以基于别人的开发板移植。但是我想知道关于 ST的标准库是已经删除了还是考虑删除？以及 LL库是已经支持了还是还在开发中？若是已经支持什么时候可以给出相关文档？

SummerGGift, 2019-05-09:
> 我以为是模板工程主要两点，一 BSP命名”f4xx“，通常的理解就是 F4系列，二，menuconfig中有 ” Device type (STM32F407VG) --->“可以配置芯片型号。而在 ”bsp/stm32“目录下都是基于具体开发板的也不存在模板工程，当然这个我可以基于别人的开发板移植。但是我想知道关于 ST的标准库是已经删除了还是考虑删除？以及 LL库是已经支持了还是还在开发中？若是已经支持什么时候可以给出相关文档？

有模板工程，请查看如下链接：
https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/libraries/templates

同时阅读如下文档：
https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/docs/STM32系列BSP制作教程.md

另外标准库以后不考虑继续支持，目前和以后主要维护基于 HAL 库的 BSP。

BernardXiong, 2019-05-09:
@TheTravels 或许可以加入一些C++的测试用例，这样后续在每个PR提交的时候，都可以到F407的开发板上进行测试。欢迎提供C++的测试用例

TheTravels, 2019-05-10:
在前面你们的回复当中并没有谈到 LL库，是否可以正面回答一下 LL库目前的开发情况？不论是正在开发还是在考虑当中。

关于 C++，我在编译的时候会出现 Thread.cpp无法编译的，错误信息为：
`
Build error: Multiple tools write to the same file. 

  File: E:\github\rt-thread\bsp\stm32f40x\build\Obj\thread.o 

  Tool 1: C/C++ Compiler, when processing E:\github\rt-thread\components\cplusplus\Thread.cpp 

  Tool 2: C/C++ Compiler, when processing E:\github\rt-thread\src\thread.c 

`
根据错误提示判断 Thread.cpp被编译成 thread.o ，而恰好源码中存在另一个 thread.c 文件，个人推测 IAR对文件明应该是不区分大小写的，而恰好两个 thread.o 又是在同一个目录，因此编译失败。这个情况对于 Semaphore.cpp同样存在。所以要么改文件名，要么改路径，再要么就不编译它直接使用内核的 C接口。

SummerGGift, 2019-05-10:
> 在前面你们的回复当中并没有谈到 LL库，是否可以正面回答一下 LL库目前的开发情况？不论是正在开发还是在考虑当中。
>
> 关于 C++，我在编译的时候会出现 Thread.cpp无法编译的，错误信息为：
> `
> Build error: Multiple tools write to the same file.
>
> File: [...]

我回答关于 ll 库的问题，目前没有考虑，只是使用 HAL 库实现驱动。

BernardXiong, 2019-05-11:
C++文件名同名是个头大的问题，后面再确认下目前scons是否已经支持了自动添加昵名文件的功能。

TheTravels, 2019-05-13:
可以确定的是目前的项目配置用 gcc编译会在 build建立子目录，按照源文件所在的相对路径建立对应的目录存放 "*.o"文件，如："build/applications/"，而 IAR不建立子目录存放(也许相应配置之后可以实现没试过)所有的 "*.o"文件都放在设定的 Obj目录，在这个基础上如果又不区分文件名的大小写那么必然会导致冲突。所以重点在于能不能让 IAR将 "*.o"文件根据对应的源文件分目录存放，这样如果冲突的话在添加源文件的时候就已经发现重名了。

BernardXiong, 2019-05-13:
原来是IAR，IAR就是异类，不支持同名文件昵名使用。这个问题无解

closed 2021-03-19 by mysterywolf