RT-Thread/rt-thread #2667: 关于 USB串口用作终端

TheTravels, 2019-05-10:
在板子初始化的时候，终端配置函数 rt_console_set_device之前是调用 rt_components_board_init函数初始化相关硬件环境，而根据内核文档该函数只调用 INIT_BOARD_EXPORT申明的函数，然而：
`
static rt_list_t class_list;

int rt_usbd_class_list_init(void)

{

    rt_list_init(&class_list);

    return 0;

}

INIT_BOARD_EXPORT(rt_usbd_class_list_init);
`
显然把 USB作为虚拟串口此时 USB并未初始化，所以，我想问下 USB串口如何作为终端使用？？？因为实际产品中如果你设计用 USB连接电脑，那么再加个 USB串口芯片显然是没有必要的，那如果要用串口进行调试显然 USB是一个很好的选择，当然如果是做开发板这点自然可以忽略。

hichard, 2019-05-10:
建议还是用USB转串口芯片吧！USB你要考虑很多问题，USB PID  VID的分配， 驱动的windows签名，否则不能安装驱动。用USB转串口芯片，一切解决，不同系统的兼容性也好。

TheTravels, 2019-05-10:
驱动和 ID都可以直接用 ST的不需要自己开发，但是除了做开发板，很多时候在实际产品中有 USB可以连接电脑都不会再考虑加 USB转串口芯片，串口更多的是接板载外设如 GPS。况且从芯片本身的角度来说既然有 USB另外再增加 USB转串口芯片来连接电脑那么 USB这个硬件是不是就有点鸡肋？然后从调试的角度讲，相比 HID等设备，USB虚拟串口用来调式程序还是比较方便的。而且 nuttx是可以支持 USB虚拟串口作为终端使用的，这个你们自己去看下源码中的配置文件就知道。我相信 USB虚拟串口要考虑很多问题，驱动、稳定性、兼容性，这些都是技术上的问题，但做产品还要考虑很多实际的问题，像成本、体积、功耗等。所以我只能说站在不同的角度看到的问题是不一样的

BernardXiong, 2019-05-11:
这部分，先熟悉RTT本身的结构，初始化，使用方式吧。

一般来说，RTT启动会有先后顺序，所以log输出在很早时就有了，rt_hw_board_init阶段就把log对外输出。

如果需要调整到USB，并且是希望一开始就向USB输出，而不是串口输出。那么可以开始有一个伪设备给到console，然后当USB枚举成功，再把console设置到USB上（上层是字符型设备，实际传输层是USB设备）。

mysterywolf, 2021-02-06:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f103-blue-pill 你好可以参见这篇readme 这篇readme详细的介绍了如何使用USB作为虚拟串口

mysterywolf, 2021-02-08:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f103-blue-pill#41-how-to-use-usb-virtual-com-as-a-console-device 你好可以参见这篇readme 这篇readme详细的介绍了如何使用USB作为虚拟串口，并将USB虚拟串口设置为console

closed 2021-02-08 by mysterywolf