RT-Thread/rt-thread #2670: 关于私有代码制作软件包

TheTravels, 2019-05-10:
根据文档《软件包开发指南》可知软件包索引是存放在”env\packages“目录下的，也就是与 env工具是放在一起的。随着我对 RT-Thread的慢慢了解，我发现基于 RT-Thread框架开发产品可以把自己的业务代码制作成软件包，这样可以方便代码的重用与后续产品的开发，我喜欢这样的一个架构。但是现在有个问题就是，业务代码显然是私有代码从公司的角度是不能够向你们提交软件包进行发布的，也就是这个软件包存放在私有服务器也许是局域网服务器。虽然 "URL"并没有限制不能为私有服务器，但是如果将私有代码制作成软件包后期加入新的软件包所有同事都需要更新 env目录，所以我在想能不能 fork之后在 RT-Thread中将私有软件包作为一个"git sub"添加到项目中来，即通过引用子模块的方式添加私有软件包，不知能否提供相关说明如何去实现？？？

liu2guang, 2019-05-10:
完全是可以的，可以添加多个软件包的源，例如在 env\packages 目录下的Kconfig：

```
source "$PKGS_DIR/packages/Kconfig"
source "$PKGS_DIR/rtpkgs_sources/Kconfig"
```

当然添加自己的子模块的fork仓库，得自己保证与上游master一致。

希望能帮助到你！

armink, 2019-05-10:
可以看下文档中心的软件包开发文档：https://www.rt-thread.org/document/site/development-guide/package/package/

里面有关于 **添加软件包源** 的介绍

TheTravels, 2019-05-10:
我觉得你们都误解了一点，我看过文档《软件包开发指南》，我的疑惑并不是如何添加软件包源，而是在 rt-thread源码的根目录机 BSP的同级目录放置一个添加软件包源，上述两个回答仍然是将新增加的软件包源存放在 env目录下。
另外 fork之后通常都是基于某一个稳定分支创建一个新的分支来开发自己的产品，并不会在原有分支上直接开发。而这个分支只是自己产品的 BSP包，真正的业务代码就是将私有代码制作的软件包。所以考虑到方便同事见的协作，将私有软件包的包源放在 rt-thread根目录作为子模块从安全方面讲也是比较符合公司的商业利益的。这些主要为了说明为什么我会有这样一个需求。所以很抱歉上述回答并没有解决我的问题。

SummerGGift, 2019-05-10:
> 我觉得你们都误解了一点，我看过文档《软件包开发指南》，我的疑惑并不是如何添加软件包源，而是在 rt-thread源码的根目录机 BSP的同级目录放置一个添加软件包源，上述两个回答仍然是将新增加的软件包源存放在 env目录下。
> 另外 fork之后通常都是基于某一个稳定分支创建一个新的分支来开发自己的产品，并不会在原有分支上直接开发。而这个分支只是自己产品的 [...]

我觉得你可以将软件包列表随便放置到哪里都行，如果你用 windows，在 env 的目录下创建快捷方式，如果你用 linux ，创建一个符号链接就好了，列表放在哪里并不重要。

只要 env 工具最终找得到就 ok。

BernardXiong, 2019-05-11:
逻辑好混乱。希望大家回答清楚你，你首先需要把自己的问题说明清楚，按照逻辑，条理性的把自己的情况说明清楚，1点2点，分列表的方式列出来。

看到一大段文字混合在一起，基本上都没看下去的欲望了

heyuanjie87, 2019-05-12:
楼主是想把自己的源放到env但是又想和主干同步而不冲突，我觉得你可以把源放在bsp只要kconfig工具能看到 是不一定要在env的

TheTravels, 2019-05-13:
我把问题重新描述一下：
1、fork RT-Thread之后在RT-Thread根目录添加自己的源，如 RT-Thread/private_packs；
2、private_packs是以 git中的子模块添加进来的，是一个单独的git版本库；
这样做的目的主要两个：
1、将自己的源与 env工具隔离；
2、private_packs作为独立版本库存放在私有或局域网服务器，确保私有代码的安全性，因为私有代码本身就不需要公开发布；
写那么多主要是想描述清楚目前的软件源管理框架与我所希望的并不一样，目前软件源都是在 env目录，如何在 RT-Thread目录添加一个软件源这才是重点！而前面的回答集中在 env目录下如何添加，所以我说并没有解决我的问题，并说你们误解了。

BernardXiong, 2019-05-13:
为什么源要放到rt-thread目录下呢，这个有些奇怪的。

所以只能说，设想的不一样。另外，这个源路径，关键在于BSP下的Kconfig怎么配置的，例如qemu-vexpress-a/Kconfig下是这样写的：

``` text
source "$PKGS_DIR/Kconfig"
```

同样的，你可以加一句：

``` text
source "$RTT_DIR/your_source/Kconfig"
```

这样就可以实现加入RT-Thread目录下的源了。

SummerGGift, 2019-05-22:
最新版本的 env 刚添加了新的功能，你可以通过如下命令：

```
set PKGS_ROOT=your_packages_source_path 
```

来设置你的软件包源目录到任意路径，想要使用该功能记得先 `pkgs --upgrade`。

closed 2021-02-08 by mysterywolf