RT-Thread/rt-thread #2681: 关于STM32外部中断

TheTravels, 2019-05-13:
根据 [STM32 系列 BSP 制作教程](https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/docs/STM32%E7%B3%BB%E5%88%97BSP%E5%88%B6%E4%BD%9C%E6%95%99%E7%A8%8B.md) 制作 BSP发现 HAL库缺少外部中断相关库。已经确认的包括：
1、v4.0.0版本F4，F1；
2、master版本F4；
其余版本未进行确认。具体如下：
按照教程步骤使用 MDK5编译错误如下：
`board\CubeMX_Config\Inc\stm32f4xx_hal_conf.h(249): error:  #5: cannot open source input file "stm32f4xx_hal_exti.h": No such file or directory`
经查看 HAL库中没有 stm32f4xx_hal_exti头文件和源文件。BSP根据 stm32f407-st-discovery移植，并没额外增加自己的代码，经查看 "v4.0.0/bsp/stm32/stm32f407-st-discovery/board/CubeMX_Config/Inc/stm32f4xx_hal_conf.h"确有如下代码：
`#ifdef HAL_EXTI_MODULE_ENABLED

  #include "stm32f4xx_hal_exti.h"

#endif /* HAL_EXTI_MODULE_ENABLED */`

非配置过程中人为增加，同时 HAL库中确实有 ”stm32f4xx_ll_exti.c“库文件，但是根据 [stm32f4xx-HAL 编译 IWG代码错误](https://github.com/RT-Thread/rt-thread/issues/2663)中的回复 LL库目前没有考虑，即不支持 LL库，而根据文档 [PIN 设备](https://www.rt-thread.org/document/site/programming-manual/device/pin/pin/)相关描述可以使用 rt_pin_attach_irq函数绑定外部中断回调函数，所以现在有几个问题：
1、HAL库中为什么没有外部中断相关代码？
2、STM32F4，F1(其他系列未具体查看)是否支持外部中断？？？是否外部中断不需要 HAL库的支持？
3、根据 ”BSP 提交规范“，只要求实现 GPIO 驱动，外部中断是否有测试以及测试是如何进行的？？？
4、我只用 CuBeMx配置了时钟，SWD，UART1，重新生成代码 ”HAL_EXTI_MODULE_ENABLED“宏就自动打开了为什么这么明显的问题都没有人发现？？？

望解答。

SummerGGift, 2019-05-13:
1. 请用 master 分支做测试。
2. LL 库说没有考虑，但是没说你不可以自己用，也没说我们把 LL 库的内容给删去了，只是实现的时候没有使用 LL 库。
3. 都支持外部中断，所有的 BSP 都经过测试可以跑起来，你先跑起来仔细看看现有的 BSP 是怎么运行的，然后再提一堆问题也不迟。

TheTravels, 2019-05-13:
我知道你认为我没有经过测试就在这里随便发表意见，上面我已经说的很明确了 master 分支我已经确认过没有 "stm32f4xx_hal_exti"相应的头文件和源文件，你自己到 HAL路径下去查看一下不就知道我是不是在这里胡说八道？你可以直接查看 github上面的内容，我目前没有给你们提交过代码，你们也不必担心我自己删除了相关文件又来这说事。而且不论github上 ”v4.0.0/bsp/stm32/libraries/STM32F4xx_HAL/SConscript“文件还是 ”master/bsp/stm32/libraries/STM32F4xx_HAL/SConscript“文件中均未使用 exti相关库，库目录下确实也没有。而实际情况是你们在 board配置都文件中有如下代码：
`/* #define HAL_EXTI_MODULE_ENABLED   */`
你们并没有开启 EXTI模块，所以你们编译不会找不到 "stm32f4xx_hal_exti.h"头文件。我对于上述问题进行过求证，列举了相应的证据，你们大可以自己查看。我也知道你们有文档说明可以使用外部中断，所以我也问了是否你们的中断实现不需要依赖 HAL库？
我知道最近我在这里提了好几个问题，也许你们觉得烦了，但是凭良心说我都是有列举证据的，你们完全可以去求证我是不是在这里胡编乱造。
最后说一点，用 V4.0.0是因为你们有文档说明这个是你们发布的稳定版本，适合用在产品中。

SummerGGift, 2019-05-13:
1. stm32f4xx_hal_conf.h 是 cubemx 自动生成的，另外确实各个固件库中都没找到  `stm32f4xx_hal_exti.h`，我开始以为是提交的时候被漏掉了，查了一下发现固件库里面就没有这个头文件，所以不知道为什么你的 cubemx 要开启这个宏。
2. 升级 cubemx 版本试一下。

TheTravels, 2019-05-13:
不是 cubemx 版本问题，我是最近下载的 5.1.0版本应该是最新的。问题应该是 HAL库的版本，我查看了一下， stm32f407-st-discovery直接打开库是 1.22.0版本，而我重新配置过的库版本为 1.24.1，但是我找过在 cubemx 软件包里面有 1.21.0和1.23.0，但是没有1.22.0版本，手动配置也无法使用 1.22.0版本。而且即使我安装 1.23.0版本我依然只能在 1.24.1和1.24.0之间选择。而且进入 ”Repository\STM32Cube_FW_F4_V1.22.0“目录查看并没有 exti相关文件，同时对应模板工程的配置文件”Repository\STM32Cube_FW_F4_V1.22.0\Projects\STM32F4-Discovery\Templates\Inc\stm32f4xx_hal_conf.h“里面并没有 ”HAL_EXTI_MODULE_ENABLED“宏的定义，注释里面都没有。所以按我的理解 1.24.1版本库把 EXTI独立出来了，所以需要这个头文件。而同时 EXIT与 GPIO合在一起的应该来说只会有 GPIIO相关文件，注释中都不应该出现 ”HAL_EXTI_MODULE_ENABLED“这个宏，在模板工程里面确实没有这个宏，为什么生成代码就会有这个我就解释不了了。
总之个人认为你们在支持 cubemx进行初始化配置的时候也许忽略了不同版本库之间的兼容性，虽然接口基本上都会是一样的，但像目前这个问题明显是对模块的划分不一样。一个把 GPIO和 EXTI统一为 GPIO，另一个则分成两个模块。这个你升级最新版本，安装最新的库测试一下就知道了。

SummerGGift, 2019-05-13:
我更新了最新版的 1.24 的库，重新生成工程，发现 `stm32f4xx_hal_conf.h` 没有打开 `HAL_EXTI_MODULE_ENABLED` 宏，同时发现整个固件库中同样也是没有 `stm32f4xx_hal_exti.h` 文件的。

由于 ST 频繁更新固件库，对结构偶尔也会有一些调整，因此同步固件库也是一项长期工作，如果你发现有更新的必要了，欢迎 PR。

TheTravels, 2019-05-13:
那这个就奇怪了，我切换到 1.24.0还是会开这个宏，而且你不放到”C:\Users\Administrator\STM32Cube\Repository\STM32Cube_FW_F4_V1.24.0\Drivers\STM32F4xx_HAL_Driver\Src“目录就是 cubemx安装库的存储目录，里面是有 exti相关文件的。或者你可以加我QQ(1875775367)我操作给你远程看一下你就清楚了。另外，因为我用的是 VET6不是VG，我是换过芯片的，或许你也可以新建一个工程用最新的库，要是你哪里再没有那真的只能说跟 cubemx的安装有关系了，因为我这个电脑是第一次安装 cubemx，没有过卸载和升级。

SummerGGift, 2019-05-13:
是我操作出了问题了，在最新的 1.24 版本里面确实有 `stm32f4xx_hal_exti.h`  这个文件，而之前使用的固件库版本里面确实没有这个头文件的。

另外我用 1.24 版本重新生成工程也并没有打开 `HAL_EXTI_MODULE_ENABLED` 这个宏，我现在不清楚什么情况下会打开这个宏。

所以问题可能是由于库版本更新导致的，如果发现了问题原因，可以一起想办法修复一下。

SummerGGift, 2019-05-13:
两种解决办法：

1. 你本地安装 1.22 版本的固件库，如下图：
[image]

2.更新 BSP 中的固件库到最新版本。

TheTravels, 2019-05-13:
有个小小建议，能不能在内核增加一个选项用来选择底层库的，通过这个选项可以选择使用的库的版本或者使用不同的库

BernardXiong, 2019-05-13:
看来版本这个事情是一个超长期的事情，只能一步步来，等到我们出对应的工具吧

nicedayzhu, 2019-05-30:
这个stm32f4xx_hal_exti.c和.h是在1.24.0版本中新增的。升级固件包到最新版本即可，目前最新的版本是1.24.1.

Soil-L, 2019-10-04:
> 两种解决办法：
> 
> 1. 你本地安装 1.22 版本的固件库，如下图：
> 
> [image]
> 
> 1. 更新 BSP 中的固件库到最新版本。

回退hal库版本应该没有办法避免开启`HAL_EXTI_MODULE_ENABLED`宏。问题的根源在于新版本cube的`stm32f4xx_hal_conf.h`模版直接开启了`HAL_EXTI_MODULE_ENABLED`宏，所以是cube的问题而非hal库。   
 
解决办法可以是把`STM32CubeMX/db/templates`路径下`stm32f4xx_hal_conf_h.ftl`文件中的`#define HAL_EXTI_MODULE_ENABLED`注释或者删除，从而保证`stm32f4xx_hal_conf_h`文件生成时不会开启`HAL_EXTI_MODULE_ENABLED`宏。

closed 2021-03-19 by mysterywolf