RT-Thread/rt-thread #2710: STM32的BSP包无法支持串口的DMA发送?

newflydd, 2019-05-22:
drv_usart.c中有如下代码：

```
#if defined(RT_SERIAL_USING_DMA)
        if(uart_obj[i].uart_dma_flag)
        {
            /* register UART device */
            result = rt_hw_serial_register(&uart_obj[i].serial,uart_obj[i].config->name,
                                           RT_DEVICE_FLAG_RDWR | RT_DEVICE_FLAG_INT_RX| RT_DEVICE_FLAG_DMA_RX 
                                           ,&uart_obj[i]);
        }
        else
#endif
        {
            /* register UART device */
            result = rt_hw_serial_register(&uart_obj[i].serial,uart_obj[i].config->name,
                                           RT_DEVICE_FLAG_RDWR | RT_DEVICE_FLAG_INT_RX
                                           ,&uart_obj[i]);
        }
        RT_ASSERT(result == RT_EOK);
    }

    return result;
}
```

以上，是利用`RT_DEVICE_FLAG_DMA_RX `这一【串口DMA接受宏】进行设备注册，但是全局搜索`RT_DEVICE_FLAG_DMA_TX`宏，发现只有定义，BSP包完全没有使用，所以也就预示着在env工具里玩出花出来都没法配置串口的DMA发送。

是否意味着官方不准备在BSP包中支持串口的DMA发送？

SummerGGift, 2019-05-22:
实际上该功能这几天正在 review，将很快加入到主分支中。

newflydd, 2019-05-23:
希望快点吧，这个应该算是重要更新吧，我在stm32_dma_config()里只看到了rx的DMA配置，完全没有关于tx的代码。

SummerGGift, 2019-05-23:
> 希望快点吧，这个应该算是重要更新吧，我在stm32_dma_config()里只看到了rx的DMA配置，完全没有关于tx的代码。

功能添加的 review 已经结束了，后面可能会逐步在各个 BSP 中添加 串口 DMA 发送的配置项了。

SummerGGift, 2019-05-24:
> 希望快点吧，这个应该算是重要更新吧，我在stm32_dma_config()里只看到了rx的DMA配置，完全没有关于tx的代码。

串口发送的代码已经合并到 master 分支了，目前可以在 [stm32f407-atk-explorer](https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f407-atk-explorer)  BSP 中使用，后续应该会逐渐应用到其他 BSP 中。

closed 2019-05-24 by BernardXiong